OctoPrint/OctoPrint #5074: [Maintenance]: new upload manager doesn't handle onBeforePrintStart callback

jneilliii, 2024-10-26:
### The problem

When starting a print from the current maintenance branch, the upload manager start print button bypasses the onBeforePrintStart callback. 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

using a plugin to validate if the callback is triggered or not

### Version of OctoPrint

OctoPrint 2.0.0.dev2449+gbdd640721.dirty

### Operating system running OctoPrint

Windows 10

### Printer model & used firmware incl. version

Virtual

### Browser and version of browser, operating system running browser

Google Chrome Version 130.0.6723.70 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241025233915.zip](https://github.com/user-attachments/files/17528875/octoprint-systeminfo-20241025233915.zip)

foosel, 2024-10-28:
I've fixed 1. but I can't reproduce 2., at least not as long as the added file type is in the `machinecode` subtree. The UploadManager checks for that, and frankly the regular load/load-and-print probably should too.

Can you check if the issue you are seeing is happening with some extension that is in the `machinecode` subtree, or some other?

jneilliii, 2024-10-28:
I'll have to retest, but it was with machinecode extension tree.

https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/d0fd4a5434043670c3fc02a2ce7020fc885be1d5/octoprint_bambu_printer/bambu_print_plugin.py#L132-L133

foosel, 2024-10-28:
I tested with a minimal plugin supporting `txt` files as `machinecode` files and was able to select and select-and-print them just fine from the upload manager:

``` python
def support_txt_machinecode(*args, **kwargs):
    return {
        "machinecode": {
            "txt": ["txt"]
        }
    }

__plugin_name__ = "Support txt machinecode files"
__plugin_pythoncompat__ = ">=3.7,<4"
__plugin_hooks__ = {
    "octoprint.filemanager.extension_tree": support_txt_machinecode
}
```

I'm confused 😅

jneilliii, 2024-10-28:
Disregard the 2nd item, apparently it was a fluke of some kind in my dev/test environment or maybe had multiple files selected or something when the button was disabled. I just tested in one of my windows installer instances and it seems to be working fine over there.

foosel, 2024-10-29:
Roger that! In that case, this can be closed as fixed.

closed 2024-10-29 by foosel