OctoPrint/OctoPrint #5076: Octoprint web page camera stream is active and downloading even when stream is not visible

timur-tabi, 2024-11-06:
### The problem

When the Octoprint webpage is loaded, there is about 500-1000 KB/s download happening of the webcam stream even when the Control tab is not visible.  That is, the Octoprint webpage is still downloading the webcam stream even if it doesn't show the actual webcam video.  

This still happens even if the web browser is minimized.  So just having the Octoprint webpage opened in any browser tab, visible or not, will force the webcam stream to be downloaded, consuming a huge amount of bandwidth.

Note that the Control tab needs to be selected at least once for the streaming to start.  Once you do that, you cannot stop the streaming without closing the Octoprint tab completely.

### Did the issue persist even in safe mode?

No, it did not persist (it's an issue with one of your installed plugins, don't report here)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

In safe mode, the webcam stream is blank even when the Control page is selected.  So webcam streaming just doesn't work in Safe Mode.

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

    OctoPi* 1.0.0cam (build 2023.07.20.144556)

### Printer model & used firmware incl. version

Prusa MK3.5s

### Browser and version of browser, operating system running browser

Firefox 132 for Linux.  Also Chrome 130, so not browser-specific.

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241105185542.zip](https://github.com/user-attachments/files/17640303/octoprint-systeminfo-20241105185542.zip)
[console-export-2024-11-5_18-59-7.txt](https://github.com/user-attachments/files/17640335/console-export-2024-11-5_18-59-7.txt)

timur-tabi, 2024-11-06:
This is the same as https://github.com/OctoPrint/OctoPrint/issues/3645, so this problem has been around for a long time.

foosel, 2024-11-07:
For the record, this is *not* the same as #3645 (at least based on the minimal amount of info provided on that ticket). You are observing this with a webcam connected whose stream won't unload, whereas the other ticket was about weird bandwidth consumption even without a webcam connected. The OP there blamed the webcam server, without realizing that a) that's not even a part of OctoPrint and b) if there's no webcam, there's no active data connection.

With that being said, I could reproduce this issue locally. Apparently something changed in how browsers handle an empty `src` attribute on an image, or something got lost when things where refactored into a plugin. An empty `src` used to stop the stream and unload it, but this is no longer happening. I've now changed the code to instead put a `data:` URL into `src` that's simply a 1px transparent gif, and that seems to work fine.

Ready for 1.11.0.

> In safe mode, the webcam stream is blank even when the Control page is selected. So webcam streaming just doesn't work in Safe Mode.

I cannot reproduce this:

[image]

If you can reliably reproduce this, then please open a new issue for that, and give us more details on your specific webcam configuration.

commit 9e0ad323 (2024-11-11): fix[classicwebcam]: use transparent gif instead of empty src for unloading

Apparently just setting an empty src stopped working a while ago.

Closes #5076
closed 2025-04-22 by foosel