OctoPrint/OctoPrint #5077: "Only show files stored on SD" doesn't show any files

timur-tabi, 2024-11-06:
### The problem

In the file browser, when "Only show files stored on SD" is selected, it doesn't show any files from the SD card attached to the printer itself. Also, the "Upload to SD" button is disabled. However, "Enable SD support" is enabled in the Features menu, and I can see and start a print from the SD via the printer's own control panel.

The terminal does show this:

Recv: ok
Send: M21
Recv: ok
Send: M155 S10
Recv: ok
Send: M20
Recv: Begin file list
Recv: STAR_0~1.GCO
Recv: End file list
Recv: ok

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.2

### Operating system running OctoPrint

OctoPi* 1.0.0cam (build 2023.07.20.144556)

### Printer model & used firmware incl. version

Prusa MK3.5s

### Browser and version of browser, operating system running browser

Firefox 132 for Linux.  Also Chrome 130, so not browser-specific.

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241105185542.zip](https://github.com/user-attachments/files/17640543/octoprint-systeminfo-20241105185542.zip)

timur-tabi, 2024-11-06:
This report is a follow-up to https://community.octoprint.org/t/only-show-files-stored-on-sd-doesnt-show-any-files/61051

This bug is similar to https://github.com/OctoPrint/OctoPrint/issues/5068

timur-tabi, 2024-11-10:
[octoprint-systeminfo-20241110093137.zip](https://github.com/user-attachments/files/17692016/octoprint-systeminfo-20241110093137.zip)

foosel, 2024-11-11:
Sorry to say, but this is a firmware bug. OctoPrint sends an `M21` to your firmware to initialize the printer's SD card. The firmware *should* be responding to that with any of

- `SD card ok`
- `Card successfully initialized`

Instead it just responds with an `ok` and leaves OctoPrint not the wiser. That OctoPrint then sends an `M20` to list the card's contents anyhow is actually a bug that I'll fix.

What you can do to work around this firmware behaviour is to enable "Always assume SD card is present" under Settings > Serial Connection > Firmware & Protocol > Firmware specific settings > Advanced options. The better solution would however be that the firmware gets fixed to report the SD card's status to OctoPrint in response to `M21` and/or automatically on initial connect, if it gets autoinitialized on boot.

foosel, 2024-11-11:
> That OctoPrint then sends an `M20` to list the card's contents anyhow is actually a bug that I'll fix.

This has been fixed via 9ed55eeac77d2ac6815b91e0db817fd3b2e9ea68.

Closing this as it's a firmware bug. A workaround has been provided.

closed 2024-11-11 by foosel

timur-tabi, 2024-11-11:
Ok, the GUI now works, but I cannot upload any .gcode files:

Send: M28 /bla.gco
Recv: ok
Send: M20
Recv: Begin file list
Recv: TRASH-~1
Recv: SKADIS~2.GCO
Recv: End file list
Recv: ok

Is it possible that my printer doesn't actually support SD upload operations?

foosel, 2024-11-11:
Looks like it. I suggest you get in touch with your printer's vendor about that. That would also explain the flaky implementation of `M21`.