OctoPrint/OctoPrint #5080: GCode Viewer seems to miss drawing some arcs

nmschulte, 2024-11-17:
### The problem

This seems a lot like #4117, but that was resolved long ago - toggling the fix there, in Chromium, has no effect; the issue is also present in Firefox.

[image]

> OctoPrint 1.10.3
> Python 3.9.2
> OctoPi* 1.0.0cam (build 2023.10.09.154319)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

OctoPi 1.0.0cam

### Printer model & used firmware incl. version

MK4 v6.1.4

### Browser and version of browser, operating system running browser

Firefox 132.0, Chromium 130.0.6723.116, Debian Linux

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241117192123.zip](https://github.com/user-attachments/files/17792071/octoprint-systeminfo-20241117192123.zip)

foosel, 2024-11-17:
Please share a gcode file with which to reproduce.

nmschulte, 2024-11-19:
[example.gcode.txt](https://github.com/user-attachments/files/17817012/example.gcode.txt)
[image]
[image]

foosel, 2024-11-19:
Thank you, with that I was able to create a minimal reproduction GCODE:

```
G28 ; home all without mesh bed level

G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion

G1 X83.869 Y113.45                                   ; 1
G1 X127.148 Y113.45 E1.26262                         ; 2
G1 X121.297 Y141.318 E.83074                         ; 3
G3 X105.568 Y154.138 I-15.789 J-3.313 E.64022        ; 4
G1 X105.568 Y154.138 F18000                          ; 5
G1 X105.508 Y153.67                                  ; 6
;TYPE:Perimeter
G1 F2400
G3 X90.176 Y141.218 J-15.665 E.62346                 ; 7
G1 X84.445 Y113.918 E.81381                          ; 8
```

The problem here was the arc command at `; 7`, `G3 X90.176 Y141.218 J-15.665 E.62346`, which doesn't have an `I` parameter. My understanding of `G2` and `G3` so far was that this was not an optional parameter for arcs (it's how the center of the arc is defined, as offset from the current location), however adjusting the code to be able to work when either `I` or `J` is missing here and assuming they are `0` in that case fixes the rendering.

[image]

The patch is done and will be part of 1.11.0. Should I need to push out a bugfix release for 1.10.x, I'll also backport this.

nmschulte, 2024-11-19:
Amazing, thank you.  Using PrusaSlicer 2.8 or so to generate that gcode.

commit 73377c57 (2024-11-20): fix[gcodeviewer]: fix rendering of G2/G3 without I or J

Closes #5080
closed 2024-11-20 by nmschulte