OctoPrint/OctoPrint #5082: Prusa Xl 5 Head Upgrade from 1.10.2 to 1.10.3 The printer profile does not generate T0 - only t1 to t4 causing a connection error on T0

bpybas, 2024-11-20:
### The problem

1. Performed an update from Octoprint 1.10.2 to 1.10.3
2. Tried to connect to my Prusa XL 5 Head printer
3. Got error message that T0 was not returning proper communications and connect failed
4. Went into the Printer Profile and Edit of Hot End and Extruder and saw that T0 was not being generated and received this warning : Please specify the nozzle offsets of the extruders relative to the first nozzle T0 if you haven't already done so in your firmware. Otherwise leave them at (0,0).
5. Have been through all the FAQ's, done a Git Hub search on Prusa XL, with no results found
6. Disabled all Plug-ins, no change
7. Rebooted in Safe mode, no change

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

Python 3.11.2

### Printer model & used firmware incl. version

Prusa XL  6.1.3

### Browser and version of browser, operating system running browser

Microsoft Edge, Win 10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241120163638.zip](https://github.com/user-attachments/files/17837337/octoprint-systeminfo-20241120163638.zip)

foosel, 2024-11-21:
And just for the record, since you seem to attribute this to the update from 1.10.2 to 1.10.3, nothing in the printer communication or handling of the printer profile was touched, as you can also see in the [changelog](https://github.com/OctoPrint/OctoPrint/releases/tag/1.10.3).

bpybas, 2024-11-22:
> I'm not sure what you mean with "does not generate T0". The printer
> profile isn't supposed to generate anything, it's only there to validate
> what you send to your printer either manually or through GCODE. You sadly
> forgot to include the exact error you are seeing.
>
> It sounds to me like you forgot to either check "shared nozzle" in your
> printer profile, or to configure your offsets. If that isn't it, then please
>
>    1. restart in safe mode
>    2. enable serial.log (Settings > Serial Connection > General > Serial
>    Logging)
>    3. reconnect to the printer
>    4. trigger the issue
>    5. create a new system info bundle and upload it here
>
> Then restart in normal mode and disable serial.log again.
>
> The bundle you provided sadly didn't contain any kind of printer
> communication.

Gina,
Thank you for the rapid response.  I went through the steps you outlined
and am unable to reproduce the error.  In going back and examining the
gcode for the print job that originally generated the gcode error, the
PrusaSlicer 2.8.1 program was the culprit.

On the Printer Profile, since there are 5 print heads I can't check the
shared nozzle option.  I was able to find the nozzle offsets in through the
Prusa XL console and those are now defined for T1-T4.

I greatly appreciate your thorough, professional approach to
problem solving.  Have donated to your efforts in the past and will
continue doing so.

Best to you,
Barry

closed 2024-11-22 by foosel