OctoPrint/OctoPrint #5085: Excessive memory usage while printing on Firefox

emrsmsrli, 2024-11-30:
### The problem

Hello, 

The problem I'm having is a case of high memory usage, most likely caused by a memory leak, if I keep the octoprint tab open for some time and there's a print job ongoing. You can see the values from Firefox internal task manager below, when I first open the tab and when it stays open for 10 minutes respectively:

[image]
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

OctoPi* 1.0.0cam

### Browser and version of browser, operating system running browser

Firefox 133.0 on Windows10

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20241130122004.zip](https://github.com/user-attachments/files/17966302/octoprint-systeminfo-20241130122004.zip)

emrsmsrli, 2024-12-01:
actually I can reproduce this without even printing anything. just keeping the tab open increases the memory usage over time

foosel, 2024-12-02:
Does it matter which tab you have open? I remember that some people reported issues when they kept the webcam stream going, which however was out of my hands as it's due to how Firefox handles the mjpg stream.

emrsmsrli, 2024-12-02:
usually it is not the active tab when this happens. maybe it is indeed related to https://github.com/OctoPrint/OctoPrint/issues/5076

foosel, 2024-12-03:
Could be. If you feel comfortable checking out the `maintenance` branch and giving that a go, that would possibly help in verifying this. Alternatively, you could remove the camera stream URL for a while and see if that helps.

foosel, 2024-12-06:
If this is indeed down to the webcam, it could be this: 

https://bugzilla.mozilla.org/show_bug.cgi?id=1280351

That would make it a browser bug, with the only thing I can do on my side is fixing the stream unload bug and suggest the use of alternative streaming formats that OctoPrint supports.

The comments in that browser ticket indicate it might be related to webcam resolution, only higher resolutions might trigger it. What do you both have configured?

nagyrobi, 2024-12-06:
> suggest the use of alternative streaming formats that OctoPrint supports.

What do you mean by that? I don't recall a way to choose or set a video streaming format.
 
> The comments in that browser ticket indicate it might be related to webcam resolution, only higher resolutions might trigger it. What do you both have configured?

I use 1920x1080, as all the camera I have support that in hardware. I also don't see any performance compromises on the CPU side of the hardware OctoPrint is running on.

emrsmsrli, 2024-12-08:
I have a stock raspi camera v2. and I can't see any resolution setting in OctoPi as well. only thing I can set is the aspect ratio, which is 16:9 at the moment

foosel, 2024-12-09:
You don't do that in OctoPrint, you have to edit the config of the webcam server (which is NOT OctoPrint) for that. For the old mjpg-streamer based camera stack, see here:

https://community.octoprint.org/t/how-can-i-change-mjpg-streamer-parameters-on-octopi/203

For the new camera-streamer based camera stack see here:

https://community.octoprint.org/t/camera-streamer-configuration-on-the-new-camera-stack-for-octopi/49950

foosel, 2024-12-09:
Also, please don't confuse OctoPrint with OctoPi, they are adjacent but not the same thing:

https://community.octoprint.org/t/what-is-the-difference-between-octoprint-and-octopi-are-they-the-same-thing/185

emrsmsrli, 2025-01-06:
cannot reproduce it anymore, I guess it's fixed on the Firefox side. closing it!

closed 2025-01-06 by emrsmsrli

foosel, 2025-06-03:
@Krojack does this also happen in safe mode?

Krojack, 2025-06-04:
> [@Krojack](https://github.com/Krojack) does this also happen in safe mode?

I'm assuming you mean OctoPrint in safe mode. I'll try this next time I do a print. It will have to be an overnight print because I need my PC during the day and it takes a few hours for memory usage to start building up.

Will report back once I get some information.

Krojack, 2025-06-09:
> [@Krojack](https://github.com/Krojack) does this also happen in safe mode?

Did a small print in safemode of an stl file that's only 12kb in size. OctoPrint's Firefox tab memory started out at 1GB usage and slowly climbed up to 6GB by the end of the print. It would sometimes jump to 8GB then back down to 5 or 6.

I modified the stl some, uploaded and started the print again and memory usage dropped to 1GB and then slowly went up just as before.

Last night I did a small print after I removed some plugins I never used. Had just some plugins loaded and it's memory usage went as high as 23GB.  Most of the plugins were pay layout ones like "UI Customizer", "OctoPrint-WideScreen", "Themeify", and "OctoPrint-Dashboard".

Firefox about:processes info (this was with some plugins enabled as listed above.
[image]

This is my total PC memory usage while running with safemode enabled. You can see when I started a new print. The total memory usage dropped from 24.86GB to 17.63GB usage.
[image]

MasterIO02, 2025-06-29:
Having the same issue. OctoPrint uses 6-7 GBs of RAM, unloads it and rapidly goes back to 6-7 GBs. This didn't happen before upgrading from 1.11.1 to 1.11.2. This lags the whole computer and makes it hard to monitor prints.

foosel, 2025-06-30:
The creation of this ticket predates the release of 1.11.0 even, so I don't see how an update from 1.11.1 to 1.11.2 could be the triggering cause.

I'm rather suspecting some weird combinations of browser versions & possibly extensions, installed plugins or the like. Still haven't been able to reproduce this at all.

So, all of you who are affected by this, please:

1. Check if that issue reproduces in [safe mode](https://faq.octoprint.org/system-info-bundle) (if not it's a third party plugin)
2. If it reproduces create a fresh [system info bundle](https://docs.octoprint.org/en/master/features/safemode.html) **from the affected browser** and post it here
3. If it happens specifically when you have a file currently selected for printing, or even currently printing, also share *that*.