OctoPrint/OctoPrint #5088: Python 3.13 install failed

fly74, 2024-12-11:
### The problem

```
(venv) op@localhost:~/venv> pip install octoprint
Collecting octoprint
  Using cached OctoPrint-1.8.7-py2.py3-none-any.whl.metadata (12 kB)
Collecting OctoPrint-FileCheck>=2021.2.23 (from octoprint)
  Using cached OctoPrint_FileCheck-2024.11.12-py2.py3-none-any.whl.metadata (1.7 kB)
Collecting OctoPrint-FirmwareCheck>=2021.10.11 (from octoprint)
  Using cached OctoPrint_FirmwareCheck-2021.10.11-py2.py3-none-any.whl.metadata (3.9 kB)
Collecting OctoPrint-PiSupport>=2022.6.13 (from octoprint)
  Using cached OctoPrint_PiSupport-2023.10.10-py2.py3-none-any.whl.metadata (2.9 kB)
Collecting argon2-cffi<22,>=21.3.0 (from octoprint)
  Using cached argon2_cffi-21.3.0-py3-none-any.whl.metadata (5.4 kB)
Collecting cachelib<0.3,>=0.2 (from octoprint)
  Using cached cachelib-0.2.0-py3-none-any.whl.metadata (2.0 kB)
Requirement already satisfied: Click<9,>=8.0.3 in ./lib/python3.13/site-packages (from octoprint) (8.1.7)
Collecting colorlog<7,>=6 (from octoprint)
  Using cached colorlog-6.9.0-py3-none-any.whl.metadata (10 kB)
Collecting emoji<2,>=1.4.2 (from octoprint)
  Using cached emoji-1.7.0.tar.gz (175 kB)
  Preparing metadata (setup.py) ... done
Collecting feedparser<7,>=6.0.8 (from octoprint)
  Using cached feedparser-6.0.11-py3-none-any.whl.metadata (2.4 kB)
Collecting filetype<2,>=1.0.7 (from octoprint)
  Using cached filetype-1.2.0-py2.py3-none-any.whl.metadata (6.5 kB)
Collecting Flask-Assets<3,>=2.0 (from octoprint)
  Using cached Flask_Assets-2.1.0-py3-none-any.whl.metadata (931 bytes)
Collecting Flask-Babel<3,>=2.0 (from octoprint)
  Using cached Flask_Babel-2.0.0-py3-none-any.whl.metadata (2.1 kB)
Collecting Flask-Login<0.6,>=0.5 (from octoprint)
  Using cached Flask_Login-0.5.0-py2.py3-none-any.whl.metadata (1.7 kB)
Requirement already satisfied: Flask-Limiter<3,>=2.6 in ./lib/python3.13/site-packages (from octoprint) (2.9.2)
Collecting flask<2.2,>=2.1 (from octoprint)
  Using cached Flask-2.1.3-py3-none-any.whl.metadata (3.9 kB)
Collecting frozendict<3,>=2.0 (from octoprint)
  Using cached frozendict-2.4.6-py313-none-any.whl.metadata (23 kB)
Collecting future<1,>=0.18.2 (from octoprint)
  Using cached future-0.18.3.tar.gz (840 kB)
  Preparing metadata (setup.py) ... done
Collecting markdown<4,>=3.2.2 (from octoprint)
  Using cached Markdown-3.7-py3-none-any.whl.metadata (7.0 kB)
Collecting netaddr<0.9,>=0.8 (from octoprint)
  Using cached netaddr-0.8.0-py2.py3-none-any.whl.metadata (4.9 kB)
Collecting netifaces<1,>=0.11 (from octoprint)
  Using cached netifaces-0.11.0.tar.gz (30 kB)
  Preparing metadata (setup.py) ... done
Collecting passlib<2,>=1.7.4 (from octoprint)
  Using cached passlib-1.7.4-py2.py3-none-any.whl.metadata (1.7 kB)
Collecting pathvalidate<3,>=2.4.1 (from octoprint)
  Using cached pathvalidate-2.5.2-py3-none-any.whl.metadata (11 kB)
Collecting pkginfo<2,>=1.7.1 (from octoprint)
  Using cached pkginfo-1.12.0-py3-none-any.whl.metadata (12 kB)
Collecting psutil<6,>=5.8 (from octoprint)
  Using cached psutil-5.9.8-cp36-abi3-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (21 kB)
Collecting pylru<2,>=1.2 (from octoprint)
  Using cached pylru-1.2.1-py3-none-any.whl.metadata (13 kB)
Collecting pyserial<4,>=3.4 (from octoprint)
  Using cached pyserial-3.5-py2.py3-none-any.whl.metadata (1.6 kB)
Collecting PyYAML<6,>=5.4.1 (from octoprint)
  Using cached PyYAML-5.4.1.tar.gz (175 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [65 lines of output]
      running egg_info
      writing lib3/PyYAML.egg-info/PKG-INFO
[... 72 lines cut ...]
note: This error originates from a subprocess, and is likely not a problem with pip.
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

newest

### Operating system running OctoPrint

Opensuse Tumbleweed

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Hi Devs, is octoprint python 3.13 ready?

fly74, 2024-12-11:
Can't get a system bundle, octoprint is not running.

cp2004, 2024-12-11:
[image]

3.13 support has not yet been implemented

foosel, 2024-12-11:
From the README:

[image]

OctoPrint 1.11.0 will support Python 3.13, first RC expected in January.

closed 2024-12-11 by foosel

fly74, 2025-02-18:
Hi @foosel I tried to install a [development environment](https://docs.octoprint.org/en/master/development/environment.html)  with python 3.13.2 and but:

```
INFO: pip is looking at multiple versions of octoprint to determine which version is compatible with other requirements. This could take a while.
ERROR: Package 'octoprint' requires a different Python: 3.13.2 not in '<3.13,>=3.7'
```
Is it still for the next release planned to support 3.13?

fly74, 2025-02-18:
Is there another git link as in the linked documentation described?

jneilliii, 2025-02-18:
https://github.com/OctoPrint/OctoPrint/releases/tag/1.11.0rc3

fly74, 2025-02-18:
```
git clone https://github.com/OctoPrint/OctoPrint/releases/tag/1.11.0rc3

Cloning into '1.11.0rc3'...
fatal: https://github.com/OctoPrint/OctoPrint/releases/tag/1.11.0rc3/info/refs not valid: is this a git repository?
```

fly74, 2025-02-18:
thx is working

[image]