OctoPrint/OctoPrint #5089: [Request] Set system time, hostname, and other system parameters.

allanonmage, 2024-12-20:
### Is your feature request related to a problem? Please describe.

Octoprint is the shell for most printers, therefore setting certain system variables like hostname and time would be appropriate for most users

### Describe the solution you'd like

A page in the settings menu to set/sync time with the local PC, set the hostname, and various other similar admin tasks.

### Describe alternatives you've considered

The only other way is to connect to the system and run a variety of tools, most of which are poorly or randomly documented.

jneilliii, 2024-12-20:
You can actually set most of this stuff as part of the flashing process from pi imager, and since OctoPrint doesn't just run on pi devices (ie windows host), I don't see this feature request being very viable for the OctoPrint project. 

https://community.octoprint.org/t/what-is-the-difference-between-octoprint-and-octopi-are-they-the-same-thing/185?u=jneilliii

allanonmage, 2024-12-20:
Nobody running Octoprint on Windows is going to need these features, since it is easy, intuitive, available in the GUI, well documented, and has not changed in where to change them on Windows.

⁣Dan Whaley, MCP, RVC-E, DMC-E, CTS
OIF II Veteran, USMC
Threema is preferred and secure: VKTCXSPJ
(770) 310-1886 cell/Telegram, txt preferred
***@***.***
LinkedIn | Github | Thingiverse | Stack Overflow​

foosel, 2025-01-06:
If someone wants to write a plugin to do this, they are welcome to, but this is not something that is going to go into OctoPrint's core. OctoPrint first and foremost is a 3d printer host, not a OS management UI.