OctoPrint/OctoPrint #5090: [Request]Connection Fix for Monoprice Mini Delta V2

DerNiceguy, 2024-12-21:
### Is your feature request related to a problem? Please describe.

Hi! I know there is a connection fix for Malyan/Monoprice printers available. It works fine with my MP Select Mini V2. It doesn't work with my MP Mini Delta V2.

### Describe the solution you'd like

After trying different things to solve the problem I use pronterface, connect to the printer followed by a reset throu the reset-button of pronterface. After that the connection will be astablished and seems to be stable.

### Describe alternatives you've considered

Can anybody code a fix witch will connect the printer followed by a reset command without interrupt the connection establishment? Maybe it is possible to combinate the fix with the existing connection fix for Malyan/Monoprice printers.

### Additional context

[image]

DerNiceguy, 2024-12-21:
[image]

DerNiceguy, 2025-01-04:
I can't recognize what pronterface is doing right to connect the MP Mini Delta V2 after the connect command followed by send a reset. No other commands, no diconnect, nothing ... connect -> reset -> connection is established successful.

foosel, 2025-01-06:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs and feature requests. See also the [Contribution Guidelines](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2025-01-06 by foosel