OctoPrint/OctoPrint #5092: [Request] List files from OctoPrint to printer to select one for printing

KimmoHop, 2025-01-02:
### Is your feature request related to a problem? Please describe.

I have been using OctoPrint to avoid SD card use and start prints remotely.
But, as I have to turn the printer on and clean the bed, I'd like to start a print right away, instead of going back to computer or having a device with browser attached to printer.

I currently have old phone viewing Mainsail on printer, but it doesn't really work. OctoPrint seems a lot lighter on the phone + having z-offset, filament change etc on printer UI is always a win (at least for me :) )

I think Portlister plugin is needed to detect printer without using OctoPrint UI?

### Describe the solution you'd like

Similar to action:prompt* except reverse: the user initiates the action on printer UI, receives a list of filenames (options) which are shown to the user and user can select one or none of those. Selection index is sent back to OctoPrint, which starts a print job with selected file.
I'm at this stage interested what the protocol between host and printer could look like? I would assume that the list should be fairly short (last N added files?) to keep printer side simple. Or some sort of paging system.
If such protocol could be plausible for both OctoPrint and Marlin, I'm slightly willing to look into it.

Minimum change would be to add new action, similar to action:start (selected file), but to start printing the newest file.

### Describe alternatives you've considered

OctoPrint with UI Customizer on phone browser nest to printer. Working, but not as easy.
Klipper/Mainsail (Fluidd is really heavy for old phone) is even less easy to use + printer display is not functional.
Start printer, go back to computer to start the print, go back to printer to see it home and start correctly (current method).
Selecting the file I uploaded doesn't work before the printer is connected.

### Additional context

Companion FR in MarlinFW: https://github.com/MarlinFirmware/Marlin/issues/27619

jneilliii, 2025-01-02:
I remember that @EvilGremlin was working on something on both Marlin and an OctoPrint plugin to support this, but haven't heard from him in a while on the OctoPrint Discord so not sure whatever happened with that.

KimmoHop, 2025-03-23:
Extending //action:start to //action:start <8.3 filename> where file name is optional + sending last 2..10 file names to printer (on connect and on file list change) + being able to store and select them in printer might be easiest.

Not sure if these could be implemented as plugin :/

If there were reverse prompt (host sends options, printer responds which was selected) this would be a lot easier ;)

KimmoHop, 2025-03-23:
Bare minimum would be to

- implement host start action in DWIN UI / ProUI (it should exist in MarlinUI)
- select newest file when printer is connected (manually or using PortLister) for host action to work

Second step could be done by implementing EventHandlerPlugin listening Connected event

KimmoHop, 2025-03-27:
FYI I implemented a plugin, that will automatically connect to printer (if set to auto-connect and serial port is defined) and after connection selects latest upload (not sure it it works with directories :/ )
And added rude //action: start to DWIN ProUI.

On the desk they seem to work, as far as it can without sensors and steppers ;)

Still I'd like to have some visibility on printer **what** will be printed beforehand...

KimmoHop, 2025-03-28:
With tricks like deleting froma and writing files to printer sd card and M118 command, providing last N files to printer and actually printing them from OctoPrint should be possible to squeeze into single plugin. No changes to printer firmware 👍

KimmoHop, 2025-04-06:
I wonder if there is any way to send comment lines to serial port? `self._printer.commands()` strips comments and I have not found good example how to use serial port (`self._printer.get_transport()`) to send - and follow responses - my own lines.
Thumbnail and some print info can be included as comment lines. It would be nice to _see_ what would be printed ;)

Sending commands has something about escaping comment with \ but does it actually work? Is escape also written to serial and is it OK for Marlin?

foosel, 2025-04-07:
I fear the only way to achieve this would be to implement your own SD file upload.

KimmoHop, 2025-04-22:
I didn't get anywhere with _direct_ (through plugin self._printer.get_transport().write(...), I did calculate line numbers and checksums, but maybe I should have used lower level serial port?) so I dropped hunting the preview.

If anyone is interested in "final" plugin, it can be found from [KimmoHop/Octoprint-AutoselectOnConnect](https://github.com/KimmoHop/Octoprint-AutoselectOnConnect) with release (zip file) for installation. It does

- automatic connection, if printer is set for that
- selecting last (uploaded) gcode file when connected to printer (for start print action)
- updating link files for last 5 gcode files to printer SD card HOST directory after connected, files uploaded or deleted (update is postponed after printing is finished, if printing)
- listening to action (sent from link file when it is "printed") and starting it from host

I'm not planning publishing it now, I'd like to get some constants into settings, but if/when I would be, is there some sort of code review procedure to maintain OctoPrint style and quality @foosel ?

Since I fulfilled my own request, I'm closing this issue. Thanks for everyone in discussion!

closed 2025-04-22 by KimmoHop