OctoPrint/OctoPrint #5094: Checksum error detection logic fails if the error message has error on its own

mbronk, 2025-01-11:
### The problem

My last print stopped suddenly due to OctoPrint sending `M112` because of the following:
```
Send: N4514 G1 X118.3210 Y109.4960 E0.0397*96
Recv: ok N4514 P0 B15
Send: N4515 G1 X117.6420 Y109.4970 E0.0641*97
Recv: Error:checksum mismatch, Last Line: 4514
Recv: Resend: 4515
Recv: ok
Send: N4515 G1 X117.6420 Y109.4970 E0.0641*97
Recv: Error:checkrum mismatch, Last Line: 4514
Changing monitoring state from "Printing" to "Error"
```

That is, there was a serial communication glitch😞 causing checksum mismatches and Marlin asking the host to resend.
However... it appears there was a serial communication error **on the 2nd error message itself** (notice the word `checkrum` not `checksum`), causing OctoPrint to fall into "unrecognized error" mode and abort the print.

It sure is environmental for me, but I think this logic of error detection:
https://github.com/OctoPrint/OctoPrint/blob/d79a0d20f3f1c7f2edb56dedda3b70267a937e65/src/octoprint/util/comm.py#L4035

may be too rigid. That is, if we're anticipating CRC errors and handling those via retransmit, shouldn't the error message detection also factor in a possible data corruption on the line?
Would it make sense to use Hamming distance to detect the "checksum" errors? Or simply allow 1 character mismatch and detect all of these regexes:  `.hecksum`, `c.ecksum`, `ch.cksum`, `che.ksum`, `chec.sum`, `check.um`, `checks.m`, `checksu.` ❓ 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Tried safe mode, but unfortunately the error is intermittent (unable to reproduce at will)

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

Ubuntu 20.04.6 @ Odroid C1+ (arm64)

### Printer model & used firmware incl. version

TwoTrees SapphirePRO running Marlin 2.1.3-b2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250111223309.zip](https://github.com/user-attachments/files/18387511/octoprint-systeminfo-20250111223309.zip)

jneilliii, 2025-01-11:
You'd then have to do the same thing for all the other ones that are in that definition? 

https://github.com/OctoPrint/OctoPrint/blob/d79a0d20f3f1c7f2edb56dedda3b70267a937e65/src/octoprint/util/comm.py#L3981-L3988

Seems too opened up at that point, and if you are getting garbage on the command line that causes the glitch you are mentioning I would consider that serial error that could be problematic. Seems best since you've identified a bug on a beta version firmware that you also report to Marlin (or whatever fork this came from) to get repaired on the firmware side.

mbronk, 2025-01-12:
> You'd then have to do the same thing for all the other ones that are in that definition?

Well, possibly. But if the logic switched to heuristics _(which may be worth considering)_, it could be done quite generically, even using the builtin [`difflib`](https://docs.python.org/3/library/difflib.html). For example:

```python
>>> from difflib import SequenceMatcher
>>> SequenceMatcher(None, "Error:checksum mismatch, Last Line: ", "Error:checkrum mismatch, Last Line: ").quick_ratio()
0.9722222222222222  #v. strong correlation
```

I'm not saying I love these kinds of heuristics. Far from it! Just wondering what systemic solution can be applied.
One of the other ideas might be a lookahead for a "Resend" request and **not halt the print** if present ❔ 

----

If we are expecting communication errors on the `TX` path (and have CRC + retries for it), I'd _guess_, a symmetric assumption (that glitches _can_ happen on the reverse (`RX`) pathway) would make some sense. And I fully acknowledge this is not an easy problem to solve, unfortunately... 😞 
My observation is just that.. everything was going smoothly (I just a few retransmits this far in the print) and error seems like it _was recoverable_. OctoPrint's safety (which I do love ❤️  and want to keep ON), was a bit oversensitive and halted the print though.

Thinking about it... I was 'lucky'  to get the glitch in the word `checksum` (_about the only one which could have caused it! 😁_) but it could equally happen on the word `error` and get completely unnoticed by this: 
https://github.com/OctoPrint/OctoPrint/blob/d79a0d20f3f1c7f2edb56dedda3b70267a937e65/src/octoprint/util/comm.py#L4012
... or any other word, and be just assumed-OK.

----

> Seems too opened up at that point, and if you are getting garbage on the command line that causes the glitch you are mentioning I would consider that serial error that could be problematic. Seems [...]

Not sure what you mean. 
Yes, the issue seems to be related to serial in general, but while one can never be certain with such intermittent issues, I'm _pretty sure_  '`checkrum`' is **not** what Marlin has sent, though. I would blame my wiring, crosstalk, interference... you name it.
While I cannot exclude any option (ex. SRAM corruption), I personally highly doubt this to be a FW bug ('`checkrum`' is nowhere to be found in Marlin's codebase, plus as you can see it has reported the error correctly just 2 lines above). My working hypothesis was that _some_ crosstalk happened at certain time in the print, affecting both `RX` and `TX` at the same time interval.

Culprit-wise, I'd probably add ferrite cores to my wire (even though it is the most shielded variant I could find), lower accelerations and what not. Or... disable OctoPrint's error checking.
While this is a one-time sighting, I wonder if it is not a sight of a systemic brittleness, lurking there in the codebase 🥷, which is why I thought I'd file it to spark a discussion whether a) it's valid, and b) anything can be done about it.

foosel, 2025-01-20:
> While this is a one-time sighting, I wonder if it is not a sight of a systemic brittleness, lurking there in the codebase 🥷, which is why I thought I'd file it to spark a discussion whether a) it's valid, and b) anything can be done about it.

It indeed is brittleness, though not in the code base but rather in the protocol. 

As you already pointed out, transmission errors might happen on both TX and RX side, yet we only have checksum capabilities built into the protocol on the TX side (from the host's perspective). That definitely is a short sight, and I'd be fine with doing some work to make OctoPrint compatible to any protocol extensions that would rectify this situation. 

I'd rather not spend time and effort on implementing code to try to guess what some garbled line of data received from the printer MIGHT have meant. That way lie dragons and it also would only be fighting a symptom (noise on the line) when it would be a better idea to fix the core issue at hand by taking measures to reduce the noise on the line. Resends take time and thus reduce throughput, you don't want garbled communication in the first place.

closed 2025-01-20 by foosel