OctoPrint/OctoPrint #5095: [RC Feedback] Feedback on the 1.11.0rc1 Release Candidate

foosel, 2025-01-28:
Please provide *general* feedback on your experience with the 1.11.0rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- The newly added Custom Control Manager plugin.
- The newly added Health Check plugin.
- The newly added Upload Manager plugin.
- The MFA-TOTP plugin using the new MFA plugin interface.
- Timelapse creation, especially with a configured rendering delay.

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

---

### Update test matrix

[image]

*Releases aren't rolled out unless all tests in the update test matrix succeed.*

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

jneilliii, 2025-01-28:
OctoEverywhere pinned sentry sdk version different than core. https://github.com/QuinnDamerell/OctoPrint-OctoEverywhere/issues/94

QuinnDamerell, 2025-01-28:
Thanks for letting me know, I will fix that ASAP.

larp-welt, 2025-01-28:
After updating to OctoPrint 1.11.0rc1, I can't upload files anymore. 

octoprint.log tells me:

```
2025-01-29 00:40:51,081 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 1473, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 882, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 880, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/lib/python3.9/site-packages/flask/app.py", line 865, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/util/flask.py", line 1643, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/vendor/flask_principal.py", line 196, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/server/api/files.py", line 692, in uploadGcodeFile
    added_file = fileManager.add_file(
TypeError: new_add_file() got an unexpected keyword argument 'user'

```
System-Info: [octoprint-systeminfo-20250129003640.zip](https://github.com/user-attachments/files/18580552/octoprint-systeminfo-20250129003640.zip)

foosel, 2025-01-29:
`new_add_file` doesn't exist in OctoPrint core, that's your "Print Time Genius" plugin falling on its face due to having overwritten an internal method and not being able to handle its new method signature.

That's why I always ask to please report bugs as *bugs*, going through the full reporting process, which would also involve testing whether the issue can be reproduced in safe mode 😅

EddyMI3d, 2025-01-29:
Hello @foosel !

I found an issue regarding uploading files from a slicer to OctoPrint via API

PrusasSlicer messages this:

> Could not connect to OctoPrint: HTTP 403: {
> "error": "You don't have the permission to access the requested resource. It is either read-protected or not readable by the server."
> }
> Note: OctoPrint version at least 1.1.0 is required.

OrcaSlicer the same.

Tried in safe_mode, same issue.

[systeminfo.zip](https://github.com/user-attachments/files/18586693/systeminfo.zip)

Tried against OctoPrint 1.10.2: that works fine.

foosel, 2025-01-29:
*Please* open a full bug report! 😅 

I'm not writing this into the release notes and the first post of the RC feedback for fun, I would say a proper full bug report is even more important for an RC than a stable release.

b-morgan, 2025-01-29:
I have an RPi 4B with OctoPrint 1.11.0rc1 Python 3.9.2 OctoPi* 1.0.0cam (build 2023.07.20.144556) but no real printer. The plugins are copied from my RPi 3B with OctoPrint 1.10.3 Python 3.7.3 OctoPi 0.18.0.

While the upgrade to 1.11.0rc1 was successful, I'm not having much luck creating a bundle as I bought myself a new computer running Windows 11 and Malwarebytes doesn't like OctoPrint yet. I'll be back with proper bug reports as soon as I can get the basics working.

foosel, 2025-01-29:
Hm, Malwarebytes seems to be tricky based on a similar issue I saw on the forums? I hope that's solvable :/

foosel, 2025-01-29:
The general issue with API keys no longer working (#5098) has now been identified and fixed (or at least I can no longer reproduce it). As that's a big issue for a lot of workflows out there I'd usually look into pushing out RC2 today, but I have a hard deadline in about an hour due to having to get where I'll be giving a talk later, so I'll see I can do that tomorrow (even though I don't like pushing releases past Wednesday - not even RCs)

b-morgan, 2025-01-29:
> Hm, Malwarebytes seems to be tricky based on a similar issue I saw on the forums? I hope that's solvable :/

What seems to work is to open OctoPrint, click on the Malwarebytes Browser Guard icon, and turn off all the "Protection for this website" sliders. I had zero luck trying to proactively add the site to the "Allow List" but this method adds the site automagically.

OdinKadaver05, 2025-01-29:
I'm having trouble with the physical buttons plugin. Worked like a charm before the Update, but since then it seems like it doesn't register the status of the GPIO pins being changed. The weirdest part of it is that GPIO-Status, a plugin for, you guessed it, checking the status of the GPIO pins still runs without issues

foosel, 2025-01-30:
Ok, as much as I'd like to push out a new RC today, it looks like #5099 is going to take a while to figure out and fix, and given that it might be a bit invasive, it's better to wait for it for a second RC to give as much testing time then as possible.

Consider me sitting here with my head feeling like it's smoking because I'm currently trying to wrap my head around the internal jinja2 templating logic again and that's oof 😅

ckuethe, 2025-01-30:
Upgrade from 1.10.0 to 1.11.0rc1 worked, connected to my printer and webcam.
Installed a new plugin successfully. Haven't tried printing with it yet, but no immediate problems.