OctoPrint/OctoPrint #5098: API requests from Apps are being rejected / invalid key

tommywienert, 2025-01-29:
### The problem

API requests are not working. In my case the applications tell the key is invalid for 
* Octodash
* Cura

A user mentioned PrusaSlicer here: https://github.com/OctoPrint/OctoPrint/issues/5095#issuecomment-2621374180

I tried recreating the keys, and even using the Global, but the keys remain invalid

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0rc1

### Operating system running OctoPrint

OctoPi 1.0.0cam

### Printer model & used firmware incl. version

Tronxy x5sa pro, marlin2 (BTT octopus)

### Browser and version of browser, operating system running browser

Firefox 134.0.2 (64-Bit), Ubuntu22.04.5 kernel 6.8.0-51

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250129124223.zip](https://github.com/user-attachments/files/18587218/octoprint-systeminfo-20250129124223.zip)

foosel, 2025-01-29:
Minimal reproduction possible with:

    curl -i -H "X-Api-Key: KEYGOESHERE" http://localhost:5000/api/printer

It doesn't create any log entries.

The issue seems to be that for some reason the permissions are now getting checked *before* `load_user_for_request` is called. I'm thinking that maybe #4968 might be the issue here, but I'm pretty sure that I tested API Keys after that commit 🤔

foosel, 2025-01-29:
Should be fixed, ready for 1.11.0rc2

XxInvictus, 2025-01-31:
I need to do some validation like restarting and safe mode but I have applied the changes you made in your fix commit and it will authenticate, but I am getting 400's when Cura is trying to upload to Octoprint. I have tried refreshing the API Token so far:
From Cura Log:
```
2025-01-31 23:52:13,072 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1608]: OctoPrintOutputDevice got an 400 error uploading to https://{redacted}:443/api/files/local
2025-01-31 23:52:13,073 - ERROR - [MainThread] OctoPrintPlugin.OctoPrintOutputDevice._onUploadFinished [1614]: OctoPrint responded with an unknown error
```

Octoprint (via Octo4a) Tornado Log:
```
2025-01-31 13:22:12 - tornado.access - WARNING - 400 POST /api/files/local (192.168.1.215) 618.92ms
```

Other Tornado Log 400's from other integration:
```
2025-01-31 10:21:34 - tornado.access - WARNING - 400 POST /api/plugin/octoapp (192.168.1.215) 95.24ms
2025-01-31 10:23:54 - tornado.access - WARNING - 400 POST /api/plugin/octoapp (192.168.1.215) 19.00ms
2025-01-31 10:26:10 - tornado.access - WARNING - 400 POST /api/plugin/octoapp (192.168.1.215) 37.67ms
```

One thing I will try that should be an easy test is going via local direct IP, both octoapp and cura are going via my public domain 🤔 
It _shouldn't_ make a difference because the .215 proxy is in the list of trusted proxies but who knows 🤷

foosel, 2025-01-31:
Please share a system info bundle

XxInvictus, 2025-01-31:
I hope you don't mind I edited the octoprint log to redact my public domain and username. Sorry I would have uploaded it earlier but I was going to wait until I had a chance to do full device reboot and/or safe mode.

[octoprint-systeminfo-20250131141434.zip](https://github.com/user-attachments/files/18619668/octoprint-systeminfo-20250131141434.zip)

XxInvictus, 2025-02-01:
Here is a brand new bundle, for this one I:
- Restarted Octoprint hardware device
- Deleted all logs
- Restarted Octoprint
- Tried to freshly setup the Cura integration
- Tried to send a print to the printer.

[octoprint-systeminfo-20250201150414.zip](https://github.com/user-attachments/files/18629201/octoprint-systeminfo-20250201150414.zip)

XxInvictus, 2025-02-01:
And another bundle, this time:
- Restart Octoprint in Safe Mode
- Try to connect Cura to Octoprint
- Try to send print from Octoprint to printer

[octoprint-systeminfo-20250201151208.zip](https://github.com/user-attachments/files/18629227/octoprint-systeminfo-20250201151208.zip)

foosel, 2025-02-03:
@XxInvictus I can reproduce this and am investigating. On first glance this is a different issue though, as it's a 400, not a 403.

Another thing however I noticed in your logs is that they're littered with

```
2025-02-01 15:02:51,214 - octoprint.util.net - ERROR - Error while trying to determine whether 192.168.1.215 is a local address
Traceback (most recent call last):
  File "/home/octoprint/octoprint-venv/lib/python3.11/site-packages/octoprint/util/net.py", line 165, in is_lan_address
    subnets = get_lan_ranges(additional_private=additional_private)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/octoprint/octoprint-venv/lib/python3.11/site-packages/octoprint/util/net.py", line 84, in get_lan_ranges
    for interface in netifaces.interfaces():
                     ^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied
```

You are using Octo4a, correct? This might be the reason this is failing. Might require some better error handling.

XxInvictus, 2025-02-03:
At least you can reproduce, I was worried for a bit there it may have been an oddity surrounding Octo4a, and yeah Octo4a is correct. Likely because from my understanding netifaces.interfaces() needs to run elevated but of course Android is too locked down 😄

XxInvictus, 2025-02-03:
I see you just did it hahaha, nevermind 😆

commit 3f45e667 (2025-01-29): fix: api keys and header auth

Misunderstanding how flask_principal and flask_login interact.

Now checking the headers for API keys and such should actually happen again before
permissions are checked.

Tested:
- regular login and UI use
- access to /api/printer via API key
- denied access when using a wrong API key

Regression probably caused by 3a61906 and for some reason not detected since.

Fixes #5098
closed 2025-02-03 by foosel