OctoPrint/OctoPrint #5101: [RC Feedback] Feedback on the 1.11.0rc2 Release Candidate

foosel, 2025-02-03:
Please provide *general* feedback on your experience with the 1.11.0rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. If you run into any bugs however, [**please open a full bug report**](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) and link it here! Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- The newly added Custom Control Manager plugin.
- The newly added Health Check plugin.
- The newly added Upload Manager plugin.
- The MFA-TOTP plugin using the new MFA plugin interface.
- Timelapse creation, especially with a configured rendering delay.

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles* in the issue tracker, not just a statement somewhere that something doesn't work! 
###
### Please respect the process and don't make the already very stressful RC phase any more difficult for me!
###
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

jneilliii, 2025-02-03:
application keys not loading #5102

jneilliii, 2025-02-03:
I suspect this change to impact Multicam plugin: <https://gist.github.com/foosel/4c57df27db637eb6136d70ab9ee6fb83#-webcamproviderplugintake_webcam_snapshot-has-gotten-its-parameters-fixed>, notified via https://github.com/mikedmor/OctoPrint_MultiCam/issues/74

b-morgan, 2025-02-03:
Upgraded 1.11.0rc1 to 1.11.0rc2 on my RPi4B (Python 3.9.2 OctoPi* 1.0.0cam (build 2023.07.20.144556). Uploaded and (virtually) printed a 3DBenchy.

foosel, 2025-02-03:
@fly74 full bug report... pleeeeeeeeeease 😭

> ### ❗👇 IMPORTANT 👇❗
> ###
> ### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug [...]

larp-welt, 2025-02-03:
Updated from RC1 to RC2, and everything works fine again.

- Upload works again!
- Spool Manager works again
- API works again with Home Assistant and OctoDash

:-)

XxInvictus, 2025-02-06:
Just adding in here, except for the Octo4a issues unrelated to this and already having an issue everything has been working! 👍

foosel, 2025-02-06:
The only thing so far found (even though the comment was deleted after asking for a proper bug report - something I'm not very amused about tbh) was the `remember_me` functionality to not work anymore. This has been fixed now and I also added an e2e test so that particular kind of bug will be found automatically in the future.

I'll give this RC another weekend and then we'll see where we are it, I'll definitely need to push out another RC though, to make sure the `remember_me` fix didn't break anything else.

fly74, 2025-02-06:
> something I'm not very amused about tbh

Sorry for deleting, but the API Key were working again, that was the main issue for me. Maybe we can start over and be friends again.

foosel, 2025-02-06:
Just don't delete any hints at there being an issue, instead just add another comment like "never mind, the reason was XYZ" or "for some reason it started working again". I WANT to follow up on and if necessary/possible fix everything found, I just need data for that beyond "ABC stopped working" (hence the repeated begging for a full bug report vs just comments here when something's broken), and deleting comments hinting at problems makes my work harder, not easier. The RC phase is the most stressful phase of every single release for me, and I need help from y'all to make OctoPrint's next release as stable as possible.

billygr, 2025-02-11:
Is the timelapse preview broken or it is only me ?
Chrome Version 132.0 MacOS

I can downloaded it and viewed locally without issues

Rest of the stuff works good meaning printing success
OctoPrint 1.11.0rc2 Python 3.7.3 OctoPi 0.18.

b-morgan, 2025-02-11:
My LulzBot TAZ 6 is connected to a RPi 3B which has a hat that controls extra fans. I running OctoPi 0.18 until I can get the hat working with a newer version. That puts me at Python 3.7.3 and I understand the warning and I'll work on it.

In the mean time, is there a way to clear the health checks? Should there be one? I understand it will come back in 30 days.

foosel, 2025-02-12:
> Is the timelapse preview broken or it is only me ?

If you mean the video player with this, it seems to be working just fine over here, in both Firefox and Chromium. In what way is it broken?

> In the mean time, is there a way to clear the health checks? Should there be one? I understand it will come back in 30 days.

There's no way to clear the health checks permanently, other than disabling the plugin altogether. And tbh I also didn't want to add one, as from my experience that will only lead to people ignoring that issue for good and then either strand on an ancient version, or worse, come complaining when they do about not having gotten a warning earlier.

What's the general opinion here, should that notification be dismissable for good, or even the check altogether?

billygr, 2025-02-12:
> > Is the timelapse preview broken or it is only me ?
> 
> If you mean the video player with this, it seems to be working just fine over here, in both Firefox and Chromium. In what way is it broken?

This is my view, play button unavailable, no preview. tried with both Chrome and Firefox (Windows 11, macOS)

[image]

foosel, 2025-02-12:
Please open a full bug report, then I can take a closer look.

Chances are this is an isolated case, but I'd still prefer to have it logged and be able to properly analyse it.

jneilliii, 2025-02-12:
> I think the check should remain and it should repeat on a schedule. I would, however, like to clear the check mark on the title bar so if any other health checks occur before the Python check reoccurs, then I'll have an indication that I should look again.

I like this idea, where the number indicator on the navbar goes away after reading, or "dismissing" the warning, but the warning still be there if opened up to view.

XxInvictus, 2025-02-13:
I also agree with the above sentiments 😄  
Generally speaking these types of warnings will be bypassed if a power user really wants.
The regular user on the other hand that sets up Octoprint to "just work" won't care about any dangers of staying on old versions or benefits from upgrading, nor will they care when they log an issue on Github with little to no information. These same people will also be the ones to clear it and never address it again despite the above things. Those who do install it to have it "just work" do tend to respond to being annoyed by messages if not at first, then 2 or 3 times later 😆

foosel, 2025-02-17:
I've now implemented a hashing logic so that all of the reported issues have a hash that can be set as "read" on the UI. The state is saved in local storage, so it's per browser (profile). The bubble will only be there as long as there are unread issues, the warning in the navbar however will be there whenever there are any issues, regardless of their reading state. The logic for the popup was not touched.

I couldn't push out a new RC last week as I was fighting with some health issues, and sadly I'm not yet fully back on my feet, but I still hope I'll be able to push out a new RC this week. It would be wonderful if I could get some feedback on whether anyone else is having issues with playing timelapses from the browser (see #5107) - I'm pretty sure this is actually a problem with Octolapse, but it would be lovely to figure this out before RC3 (so ideally today or tomorrow the latest).

jneilliii, 2025-02-17:
Default timelapse preview seems to be working fine for me. 

[image]

billygr, 2025-02-18:
> playing timelapses from the browser (see [#5107](https://github.com/OctoPrint/OctoPrint/issues/5107)) - I'm pretty sure this is actually a problem with Octolapse, but it would be lovely to figure this out before RC3 (so ideally today or tomorrow the latest).

Confirmed , disabling Octolapse solves it, will update the relevant case

foosel, 2025-02-18:
1.11.0rc3 has just been released, new ticket is #5111

closed 2025-02-18 by foosel