OctoPrint/OctoPrint #5103: Gcode viewer doesn't show the entire layer

Zebski, 2025-02-03:
### The problem

Gcode Viewer doesn't show the entire layer. It processe 1 bit of it then procedes to the next "layer" but in reality it's still the same layer. Something seems to make it think there was a layer change, when there wasn't.
Allowing to show next layer doesn't fix it as it's not a real layer change

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Gcode viewer is a plugin so I'm not sure I can test it in safe mode?

### Version of OctoPrint

OctoPrint 1.10.3 Python 3.9.2 

### Operating system running OctoPrint

OctoPi* 1.0.0cam (build 2024.06.18.101629)

### Printer model & used firmware incl. version

Ender 3 Pro, FW 1.3.1 with BL Touch

### Browser and version of browser, operating system running browser

Windows 11, Chrome Version 132.0.6834.160 (Officiel version) (64-bit)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [x] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [x] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octopi.local-1738595433502.log](https://github.com/user-attachments/files/18644151/octopi.local-1738595433502.log)

[octoprint-systeminfo-20250203161217.zip](https://github.com/user-attachments/files/18644181/octoprint-systeminfo-20250203161217.zip)

[CE3PRO_Blue_LockSpool3_1kg_SUNLU_PETG.zip](https://github.com/user-attachments/files/18644299/CE3PRO_Blue_LockSpool3_1kg_SUNLU_PETG.zip)

foosel, 2025-02-03:
> Gcode viewer is a plugin so I'm not sure I can test it in safe mode?

It's a bundled plugin, and bundled plugin aren't deactivated in safe mode, so please test in safe mode, just in case.

Also, for the record, I currently have my hands full with the release candidates for 1.11.0, so it might be a while until I can look into this (and any fix will have to wait for at least the first bug fix release after 1.11.0 stable goes out, 1.11.0 is now already frozen apart from regression fixes)

Zebski, 2025-02-04:
Thanks, - I'll test in safe mode and see if the problem persists and report
that to the case. I can see Pretty gcode counts a lot more layers, so there
is definitely something fishy about it.
Number of layers in the left side panel and on the printer shows correctly,
so it's just a minor inconvenience, but it just started to bug me that it
wasn't showing the correct visualization.

Regards
Bjarne Zeb Jørgensen

Den man. 3. feb. 2025 kl. 17.23 skrev Gina Häußge ***@***.***

foosel, 2025-02-04:
It's probably something very specific to the output of your slicer, so if you could also share what you are using to generate that gcode it would help. Especially any new or less common settings.

Zebski, 2025-02-04:
Ok, I’ll export the settings ftom Cura in a 3mf file

foosel, 2025-02-04:
Do you have scarf joint enabled? If so this is likely a duplicate of #5039

foosel, 2025-02-05:
Looking at this a bit closer in PrusaSlicer's GCODE viewer, it looks indeed like scarf joint, marking as duplicate of #5039.

closed 2025-02-05 by foosel

Zebski, 2025-02-06:
Yes, I have enabled scarf joint in Cura. Thank you! I was going nuts about
what had changed...
Mvh
Bjarne Zeb Jørgensen

Den ons. 5. feb. 2025 kl. 12.07 skrev Gina Häußge ***@***.***

Zebski, 2025-02-06:
You're the best - just signed up for a monthly payment. The work you do is
awesome!

Mvh
Bjarne Zeb Jørgensen

Den ons. 5. feb. 2025 kl. 12.07 skrev Gina Häußge ***@***.***