OctoPrint/OctoPrint #5104: Improve error handling/add work around for netifaces issues under octo4a

foosel, 2025-02-04:
### Problem

> Another thing however I noticed in your logs is that they're littered with
> 
> ```
> 2025-02-01 15:02:51,214 - octoprint.util.net - ERROR - Error while trying to determine whether 192.168.1.215 is a local address
> Traceback (most recent call last):
>   File "/home/octoprint/octoprint-venv/lib/python3.11/site-packages/octoprint/util/net.py", line 165, in is_lan_address
>     subnets = get_lan_ranges(additional_private=additional_private)
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "/home/octoprint/octoprint-venv/lib/python3.11/site-packages/octoprint/util/net.py", line 84, in get_lan_ranges
>     for interface in netifaces.interfaces():
>                      ^^^^^^^^^^^^^^^^^^^^^^
> PermissionError: [Errno 13] Permission denied
> ```
> 
> You are using Octo4a, correct? This might be the reason this is failing. Might require some better error handling. 

 _Originally posted by @foosel in [#5098](https://github.com/OctoPrint/OctoPrint/issues/5098#issuecomment-2630393637)_

### Solution

Let's figure out how many affected instances are there first by adding `platform.platform()` to the environment report and the anonymous usage tracking.

Then based on that data we'll have to decide on whether we can maybe just disable interface address related functionality on that restricted platform, or if we have to find a platform specific workaround.

foosel, 2025-02-04:
Quoting the excellent detective work done by @XxInvictus on [Discord](https://discord.com/channels/704958479194128507/705047010641838211/1336329205759213590):

> Hey foosel it's XxInvictus from Github, I just wanted to share that I dug a bit deeper into the side-issue that was "didn't really seem to be an issue but resulted in log errors" with Octo4a and being an Alpine sandbox within Android it not having the necessary permissions to return `netifaces.interfaces()` (and also subsequently discovered `netifaces.ifaddresses(interface)`).
> 
> First thing was trying to find an alternative fallback method that could be used in-case of failure, TLDR there doesn't seem to be a good method that works around the implementation of Octo4a. Will get into that a bit after.
> 
> Second thing was trying just to throw some error-handling in so it fails gently but I am not sure of the repurcussions of this one as the dependencies on this go all the way back to the zeroconf implementation.
> 
> Finally, looping back to the first, most Python native alternatives I tried, like `socket.getaddrinfo()`  on the hostname, resulted in only getting loopback addresses returned. The only real alternative that I could find that will return all of the information used by the dependencies of the two functions (mainly IP and Netmask) is a couple of ugly methods.
> 
> Bearing in mind the below are ONLY if the netifaces method fails with `PermissionError`, could potentially go a step further and check for `linux` from `platform.system()` or `android` in `platform.release()`.
> 
> Method A - Parse all information from the `ifconfig -a` OS command
> 
> - Cleanest to get all information in one place
> - Only really suitable for this specific environment
> - I hate using OS commands within PY personally
> 
> Method B - Parse only interface names from `ifconfig` then use python `ioctl` system calls to retrieve the IP and Netmask
> 
> - I haven't yet been able to find a native/alternative way to get the interface names without ifconfig, hopefully more digging will come up with something
> - Retrieves only the bare minimum from ifconfig and uses ioctl system calls to get the rest
> 
> Rough example you can easily fill in the blanks on what is implied:
> 
> ``` python
> for interface_name in interfaces:
>     with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as s:
>         ipadd = socket.inet_ntoa(fcntl.ioctl(s.fileno(), 0x8915, struct.pack('256s',bytes(interface_name, 'utf-8')))[20:24])
>         netmask = socket.inet_ntoa(fcntl.ioctl(s.fileno(), 0x891b, struct.pack('256s',bytes(interface_name, 'utf-8')))[20:24])
> ```
> 
> Some of the failed runs include:
> 
> - Using psutil net_if_addrs(), results in a segmentation fault
> - Using ctypes to tap into libc, segmentation fault
> - Trying to list interfaces using sockets.if_nameindex(), results in No buffer space available

[...]

> For posterity sake even if it is never used, this is the Python that allowed fetching all interfaces without using `ifconfig` that I at least know works on Unix and in the Android-Alpine jail. I haven't broken it down yet but given it is using the same concepts as that I used above to get the IP and Netmask it could probably be adapted to get [(name, ip, netmask),]
> ```
> >>> import socket
> >>> import fcntl
> >>> import struct
> >>> import array
> >>> import sys
> >>>
> >>> def all_interfaces():
> ...     is_64bits = sys.maxsize > 2 ** 32
> ...     struct_size = 40 if is_64bits else 32
> ...     s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
> ...     max_possible = 8  # initial value
> ...     while True:
> ...         _bytes = max_possible * struct_size
> ...         names = array.array('B')
> ...         for i in range(0, _bytes):
> ...             names.append(0)
> ...         outbytes = struct.unpack('iL', fcntl.ioctl(
> ...             s.fileno(),
> ...             0x8912,  # SIOCGIFCONF
> ...             struct.pack('iL', _bytes, names.buffer_info()[0])
> ...         ))[0]
> ...         if outbytes == _bytes:
> ...             max_possible *= 2
> ...         else:
> ...             break
> ...     namestr = names.tobytes()
> ...     ifaces = []
> ...     for i in range(0, outbytes, struct_size):
> ...         iface_name = bytes.decode(namestr[i:i + 16]).split('\0', 1)[0]
> ...         iface_addr = socket.inet_ntoa(namestr[i + 20:i + 24])
> ...         ifaces.append((iface_name, iface_addr))
> ...     return ifaces
> ...
> >>> all_interfaces()
> [('lo', '127.0.0.1'), ('wlan0', '192.168.1.190')]
> ```

XxInvictus, 2025-02-06:
Just throwing down a quick thought if the market share of Android/other similar devices is too small to warrant putting in a proper alternative and it is a nuisance to do purely with error-handling.  
Given the data being collected by this function is pretty simple it might be worth adding it to the settings to allow a user to configure the IP and Netmask which would then serve to bypass the checks and just use those values.

XxInvictus, 2025-02-06:
That makes sense, I am an idiot 😆 my days are a blur I thought for a second I was just misremembering and it was before RC2 🤦

kohend, 2025-05-19:
Using settings would be a partial solution since zeroconf also gets this information, and it's needed for plugins like discovery.

XxInvictus, 2025-05-20:
My time has been taken up by a new puppy so the printer took a back-seat but just in case there is some cross-contamination of Issues here AFAIK `ifaddr` still won't work for Android due to the same permission issues that are encountered with `netifaces`. I am 95% sure I tried `ifaddr` during my testing and it was still captured by the Android "permission jail" with the result being that the only way to get the information in it's entirety was unfortunately through pulling it directly from the `io` device using sockets and ioctl in my above examples.

foosel, 2025-05-27:
Slightly off-topic, but happy to get a picture of the puppy - I absolutely *love* dogs!

And yeah, I also am skeptical that ifaddr solves this, but at the very least one less dependency now!

XxInvictus, 2025-07-14:
Not looking promising on ifaddr change hahaha, I would be very interested if they do accept the PR because currently it doesn't look like they use subprocesses anywhere and something like parsing the output in-full from ifconfig is prone to fault if the output changes format in any way. But hey, if they choose to accept it I am not going to argue with a dependency gone 😛 

Obligatory puppy tax @foosel 😛 
Soon after we got him:
[image]
Recent:
[image]