OctoPrint/OctoPrint #5106: [Request] When printing, pre-pend %complete to page title

STR4NG3RdotSH, 2025-02-10:
### Describe the solution you'd like

[Request] When printing, pre-pend % to page title.

For example, currently in my browser, I have Octoprint.local open in a tab and the tab is labeled `OctoPrint`. I think it would be a simple and greatly appreciated feature if, during printing say the job is 69% done, the title instead would read: 
`69% - OctoPrint`
or 
`69% MyCurrentlyPrinting.STL - OctoPrint`

Or this could even be a toggleable option in the settings, or some custom wildcard to use in the already existing custom title options.
In any case, if implemented, I think the % should definitely be the start of the title so its always visible despite tabs being tiny from user having tons of tabs open.

Just an idea. Since I know my lazy butt would appreciate it, there's bound to be others out there who feel the same.

cp2004, 2025-02-10:
You can do this with a plugin eg. Display Layer Progress. It has a lot of configuration as to what metrics you can see.

STR4NG3RdotSH, 2025-02-11:
I didn't even consider checking if there was a plugin that could do this! I'll check that plugin out, thanks :)

I do feel this could still be a valuable official feature, just the same.

jneilliii, 2025-02-11:
There's even a plugin for this specific request, that may not be as bulky. https://plugins.octoprint.org/plugins/progress_title/

jneilliii, 2025-02-11:
Or this one: https://plugins.octoprint.org/plugins/tabinfo/

foosel, 2025-02-17:
I don't see a point in reimplementing functionality that already is provided by the community as plugins, as long as those plugins are not abandoned, so I'm closing this as a solution exists.