OctoPrint/OctoPrint #5107: OctoPrint 1.11.0rc2 TimeLapse Preview unavailable (black popup, controls unavailable)

billygr, 2025-02-13:
### The problem

Issues started when upgrading to 1.11.0rc2

Step to reproduce it
Clicking on any available timelapse 
pop up opens
Preview is not available

JS errors on  Please use the webcam system introduced with 1.9.0, the webcam_rotate90 config setting is deprecated and will be removed in a future release. Stacktrace: Error

Could be related

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

OctoPrint 1.11.0rc2 Python 3.7.3 OctoPi 0.18.0

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

macOS Chrome Version 133.0.6943.54 (Official Build) (arm64)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [x] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250213125309.zip](https://github.com/user-attachments/files/18782658/octoprint-systeminfo-20250213125309.zip)

[js_errors.txt](https://github.com/user-attachments/files/18782708/js_errors.txt)

[image]

foosel, 2025-02-13:
Can you share your JS error console output with safemode enabled? I want to make sure I'm not hunting down third party plugin warnings/errors here.

billygr, 2025-02-13:
[js_output_safemode.txt](https://github.com/user-attachments/files/18783166/js_output_safemode.txt)

foosel, 2025-02-13:
So all of the errors reported on the browser console are from third party plugins. As you said that the issue also happens under safe mode (correct?) that's not it.

I'll see if I have luck with reproduction on macos via browserstack, but that's about the only idea I have right now, it's working just as it should here in both Firefox and Chrome, and I also can't say that I touched the viewer in any way as far as I remember.

billygr, 2025-02-13:
Issues also happens with safe mode with both Firefox and Chrome (and with Windows 11 using Chrome/Firefox)

billygr, 2025-02-13:
I don't know if it helps but i think i traced it to the following:

videos with the following filename format:

FileName.aw_20250209221618.mp4

Meaning with 2 dots in the filename

Went back 2-3 years and the ones with a single dot work (in the preview as expected)

billygr, 2025-02-13:
Second dot is coming from ArcWelder uploading an affected timelapse

https://github.com/user-attachments/assets/c36fe364-32f5-4348-8dfc-f1e0bebc5bf6

foosel, 2025-02-13:
Well, whatever it is, it's something that's caused by Arcwelder then, because even the embed in the ticket above isn't working:

[image]

foosel, 2025-02-13:
Ok, even downloading it and trying to watch it in vlc doesn't work though, so just to make sure this isn't caused by something here on GitHub's event, can you please upload it here: https://cloud.octoprint.org/s/rbtApHm4fr8nSog

foosel, 2025-02-13:
Ok, that was my mistake when downloading via curl, I can now play it in VLC. Threw it on my dev version and it won't play. So something's wrong with those files getting created on your end. Did you change the `ffmpeg` parameters? Default should be `{ffmpeg} -framerate {fps} -i "{input}" -vcodec {videocodec} -threads {threads} -b:v {bitrate} -f {containerformat} -y {filters} "{output}"` and video codec `libx264`.

[image]

`ffprobe` shows this codec info for your file:

```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'test.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    encoder         : Lavf58.20.100
  Duration: 00:00:05.47, start: 0.000000, bitrate: 5064 kb/s
  Stream #0:0[0x1](und): Video: mpeg4 (Simple Profile) (mp4v / 0x7634706D), yuv420p, 640x480 [SAR 1:1 DAR 4:3], 5061 kb/s, 60 fps, 60 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
```

And this is one of mine, created yesterday under 1.11.0.dev and working:

```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'e2e-test_20250212181123.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf60.16.100
  Duration: 00:00:01.08, start: 0.000000, bitrate: 3895 kb/s
  Stream #0:0[0x1](und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt470bg/unknown/unknown, progressive), 1280x720, 3886 kb/s, 25 fps, 25 tbr, 12800 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 libx264
```

billygr, 2025-02-13:
Thank you for your hint, it checked all the files with ffprobe and there is a mix between h264 and mpeg4

As for the settings i am using the octolapse plugin however i double checked that the command line for FFMPEG is the one you provided including the libx264

Could be the case of a silent upgrade on the settings of octolapse

foosel, 2025-02-13:
If you are usually running your timelapses via octolapse that could explain it. 

Could you enable the virtual printer and run some short print through it, using your usual timelapse settings, then share another system info bundle? I'd hope there'd be the rendering command in there and we could compare it then with what OctoPrint should be using.

foosel, 2025-02-17:
Sadly I can't find the rendering command in there. Maybe it only gets logged with debug output. In any case, I'm pretty sure this is caused by Octolapse rendering timelapses differently than core. Please disable Octolapse, do a short print with OctoPrint's own timelapse enabled and see if that then plays back in the browser. My guess is it will do just fine, and this is actually a third party plugin issue that's probably been going on for a while now.

billygr, 2025-02-18:
Not an Octoprint issue but an Octolapse issue, with a virtual printer and the plugin disabled it renders correctly

foosel, 2025-02-18:
Perfect, then this should best be reported to Octolapse (though tbh I'm not sure how actively maintained it is these days). I quickly looked through the source to see if I could easily figure out where the issue lies, but so far no win.

Closing this as it's not an issue with OctoPrint itself, but rather a third party plugin issue.

closed 2025-02-18 by foosel