OctoPrint/OctoPrint #5109: `octoprint.systemcommands.SystemCommandManager`'s `has_*()` returns True if command was once set, and then blanked out.

cnf, 2025-02-16:
### The problem

`octoprint.systemcommands.SystemCommandManager`'s `has_*()` returns True if command was once set, and then blanked out.

so if you set `Restart OctoPrint` in the UI to any value and save, and then blank out said field, and save again, `has_system_restart_command()` will now return True

I suspect this is because the settings don't get removed, but just blanked.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

I am developing in a module, and I need the module code to run to actually test this.

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

Linux, in a dev env in VSCode.

### Printer model & used firmware incl. version

virtual printer

### Browser and version of browser, operating system running browser

curl

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250216144229.zip](https://github.com/user-attachments/files/18814851/octoprint-systeminfo-20250216144229.zip)

foosel, 2025-02-17:
Thanks for spotting this, will see if I can push the fix for that still into 1.11 (even though that's already on RC level). Fix is small enough to risk that.

foosel, 2025-02-18:
Closing as this has been fixed and rolled out in 1.11.0rc3.

commit 1ac8e4cc (2025-02-17): fix: detect empty system commands

Fixes #5109
closed 2025-02-18 by foosel