OctoPrint/OctoPrint #5111: [RC Feedback] Feedback on the 1.11.0rc3 Release Candidate

foosel, 2025-02-18:
Please provide *general* feedback on your experience with the 1.11.0rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. If you run into any bugs however, [**please open a full bug report**](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) and link it here! Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- The newly added Custom Control Manager plugin.
- The newly added Health Check plugin.
- The newly added Upload Manager plugin.
- The MFA-TOTP plugin using the new MFA plugin interface.
- Timelapse creation, especially with a configured rendering delay.
- Remember me functionality

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles* in the issue tracker, not just a statement somewhere that something doesn't work! 
###
### Please respect the process and don't make the already very stressful RC phase any more difficult for me!
###
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

- https://github.com/OctoPrint/OctoPrint/issues/5107 - actually an issue with Octolapse, should be reported upstream

### Issues needing analysis

b-morgan, 2025-02-18:
Rpi 4B (OctoPrint 1.11.0rc3 Python 3.9.2 OctoPi* 1.0.0cam (build 2023.07.20.144556)) upgraded successfully.

RPi 3B (OctoPrint 1.11.0rc2 Python 3.7.3 OctoPi 0.18.0) has the announcement but is very reluctant to offer me the update. Restarted OctoPrint, rebooted the system. Off to generate a bug report...

foosel, 2025-02-18:
As mentioned in #5112, that was indeed just a cache.

b-morgan, 2025-02-18:
I was able to clear the Health Check count, the icon is still there, and I guess I'll have to wait 30 days to see if it comes back.

Should release candidates have a shorter recheck interval?

foosel, 2025-02-18:
No, should be the same. You can manually delete the cookie however via the dev console (Storage tab) to verify the popup notification returns. Note that only THAT will return, not the red bubble on the icon.

b-morgan, 2025-02-18:
I get in enough trouble already so manually deleting cookies is above and beyond. It does, however, make me hungry so I'm off the eat a RL cookie 😁

fly74, 2025-02-18:
> * Remember me functionality

I can confirm it is working again.

AndKe, 2025-02-19:
RC2 worked
RC3 fails to connect to printer:

[image]

rebooted octoprint, the printer,   I selected the serial port to show it's existence, normally all is set to auto and just works. need to roll back.

AndKe, 2025-02-19:
also: 

[octoprint-systeminfo-20250219142829.zip](https://github.com/user-attachments/files/18869892/octoprint-systeminfo-20250219142829.zip)

foosel, 2025-02-19:
@AndKe I didn't touch anything related to printer communication between RC2 and RC3, so I doubt this is due to the update. Still: make sure to enabled serial.log, try again, create a system bundle, create a full bug report including that.

AndKe, 2025-02-19:
@foosel  - please note that I posted the log bundle right after.
  I restarted the printer (toggled the PSU several times, restarted the octiprint and rebooted the pi)
nothing helped.
Finally I did:
 ~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint/releases/download/1.11.0rc2/octoprint-1.11.0rc2.tar.gz
(and restarted the service)  - and boom ! - everything works again. 

maybe this was some one in  a million craziness :)   - or maybe you get more reports like this.  - I may retry later, need to I got the "need to print" now.  :)

AndKe, 2025-02-19:
once the print is finished, I'll re-install RC3 and give you what you need.

AndKe, 2025-02-19:
@foosel After again updating to rc3  I am unable to reproduce the problem anymore.  I even restarted the 3D printer, after the upgrade, connected, then restarted OctoPi too - everything just works. 
That was very odd, and definitely coincided with the upgrade, but it must have been some glitch/corruption/temporary issue as it is clearly gone now.  Sorry for wasting your time - I thought it could be valuable feedback.

foosel, 2025-02-19:
No worries, and no time wasted. Glitches can also be issues. The RC phase is there to find everything that can be found, which is why I also want to analyse stuff like that (if it stays reproducible). 

I prefer stuff that happens to not be an issue to be reported over stuff that is an issue not being reported by far 😬

jneilliii, 2025-02-27:
Small issue with health check plugin under certain conditions: https://github.com/OctoPrint/OctoPrint/issues/5115

pooh22, 2025-03-06:
I have an unclear problem that a print just halts in mid print, in the first case it stopped on a corner and never continued, I restarted octoprint (1.11.0rc3 on an Rpi 3) and then the print was gone from the interface.
The second time happened just now and the print continued after a minute or so (I think it may have skipped a command to go to the next layer from the noise of the nozzle bouncing on the infill ;-)

The first time I was looking at the gcode preview while it was printing and it was loading a long time, may have caused a problem sending the gcode over the serial line on time?
The second time it just happened, although I was upgrading home assistant on a different device, but in a different tab of the same browserwindow. I can't imagine the browser will link the two things, but octopi is known in home assistant.

The print is still running, but it seems jerky (like sometimes normal speed, sometimes hesitant, though fine for now). It may be in the gcode, that it is slower on overhangs and stuff, but it "feels" like it may be something else.

pooh22, 2025-03-06:
now again, it is like the gcode commands almost stop or need to go through a 300 baud modem, they don't entirely stop, just one every 10 seconds or more

jneilliii, 2025-03-06:
> now again, it is like the gcode commands almost stop or need to go through a 300 baud modem, they don't entirely stop, just one every 10 seconds or more

Please follow instructions to take action/investigate your issue in separate thread vs cluttering this one up.

[image]

pooh22, 2025-03-06:
I'm sorry for cluttering this thread up! I didn't have time to look thoroughly, looking at existing bugs, it looks a lot like https://github.com/OctoPrint/OctoPrint/issues/2647, however I've never seen it before. I didn't have it before I re-installed (octopi most recent to remove the python version warning) and restored from backup (of 1.11.0rc3). But I don't have time to go deeper and investigate. That's also why I didn't file a separate report (has anyone mentionioned that the hurdle is very high to even file a report? I'm just a user (and linux sysadmin, so I know my open source stuff), I have 3 kids and a house, this is about the most help I can offer (besides paying foosel to keep up the good work). I'll keep quiet for now, perhaps I should switch to prusalink?

foosel, 2025-03-11:
1.11.0rc4 has been released, new feedback ticket is here: #5120

closed 2025-03-11 by foosel