OctoPrint/OctoPrint #5112: OctoPI 0.18.0, OctoPrint 1.11.0rc2 does not offer the update to rc3

b-morgan, 2025-02-18:
### The problem

OctoPi 0.18.0, OctoPrint 1.11.0rc2 has an announcement for rc3 but does not offer the update.

Since I successfully updated my RPi4B, I believe that eliminates the possibility that the file(s) are not available.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0rc2

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

LulzBot TAZ 6, Marlin 2.0.9.0.13

### Browser and version of browser, operating system running browser

Windows 11, Google Chrome Version 132.0.6834.197 (Official Build) (64-bit)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250218082158.zip](https://github.com/user-attachments/files/18847588/octoprint-systeminfo-20250218082158.zip)

[image]

b-morgan, 2025-02-18:
"Check for updates" did nothing, "Force check for updates" worked.

foosel, 2025-02-18:
Then it is all working as intended (and since 2015) and it was indeed the cache.

closed 2025-02-18 by foosel