OctoPrint/OctoPrint #5115: [1.11.0rc3] Health Check Plugin mime types error when not bundled/minified

jneilliii, 2025-02-27:
### The problem

With config.yaml set to not bundle and minimy webassets there are errors in developer tools console related to loading files. Not sure if it's one setting are the other, haven't had time to test. 

```
devel:
  webassets:
    bundle: false
    minify: false
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0rc3

### Operating system running OctoPrint

Windows 10 (my installer)

### Printer model & used firmware incl. version

Virtual

### Browser and version of browser, operating system running browser

Google Chrome Version 133.0.6943.141 (Official Build) (64-bit)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [x] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]

[octoprint-systeminfo-20250227114458.zip](https://github.com/user-attachments/files/19013083/octoprint-systeminfo-20250227114458.zip)

foosel, 2025-03-04:
That's not an issue with the CSS type, that's a case of a wrong URL being injected into the assets: 

```
$ curl --head --silent http://localhost:5000/plugin/health_check/static/css%5Chealth_check.css | grep -i content-type
Content-Type: text/html; charset=utf-8
X-Content-Type-Options: nosniff
$ curl --head --silent http://localhost:5000/plugin/health_check/static/css/health_check.css | grep -i content-type
Content-Type: text/css; charset=utf-8
X-Content-Type-Options: nosniff
```

I could imagine that this is a side effect of the auto-escape and you just found a corner case where I forgot to mark something as safe in the template.

Will look into it.

foosel, 2025-03-04:
Hm.... Alas, if only I could reproduce it.

Disabled both bundling and minification (the latter was already the case):

``` yaml
devel:
  webassets:
    minify: false
    bundle: false
```

Reloaded. Refreshed cache. Tried a temporary container tab. Works everywhere.

Checked via curl. URLs look fine there too:

```
$ curl --silent --header "X-Api-Key: REDACTED" http://localhost:5000/ | grep -oP 'src="/plugin/health_check/static/[^"]+"'
src="/plugin/health_check/static/clientjs/health_check.js"
src="/plugin/health_check/static/js/health_check.js"
$ curl --silent --header "X-Api-Key: REFACTED" http://localhost:5000/ | grep -oP 'href="/plugin/health_check/static/[^"]+"'
href="/plugin/health_check/static/clientjs/health_check.js"
href="/plugin/health_check/static/js/health_check.js"
href="/plugin/health_check/static/css/health_check.css"
href="/plugin/health_check/static/css/health_check.css"
```

That's OctoPrint 1.11.0rc4.dev3+g8a255faba under Python 3.11.2. Tested in Firefox and Chromium (browser shouldn't matter here, from the shape of the error this must be something that's already wrong in the HTML response).

Can you try checking the response with curl too? (see above on how to)

jneilliii, 2025-03-04:
This is the result of those commands with 1.11.0rc3.

[image]

Maybe the issue was fixed in https://github.com/OctoPrint/OctoPrint/pull/5116, which I assume is what your rc4 dev version is running?

jneilliii, 2025-03-04:
Yeah, I updated to staging/maintenance branch (OctoPrint 0+unknown.g348dc1ed) and it still does it. SoftwareUpdate plugin doesn't, only the health check plugin seems to be impacted. 

[image]

foosel, 2025-03-04:
Ooooooh, wait. `%5c` isn't a slash, it's a backslash. I might have an idea what's happening here. Windows vs auto fetch of assets.

foosel, 2025-03-04:
The above commit should hopefully fix this, can you try a current checkout of `staging/maintenance`?

foosel, 2025-03-04:
Confirmed as fixed on Discord, thanks! Fix ready for 1.11.0rc4

commit e81bdf08 (2025-03-04): fix: convert \ to / in auto-discovered asset paths

Because windows...

Fixes #5115