OctoPrint/OctoPrint #5121: [1.11.0rc4] error retrieving data for sd card files in log

jneilliii, 2025-03-15:
### The problem

Not sure if this is a red herring or not (plugin related), but I am seeing an error in 1.11.0rc4 log that I don't have in 1.10.3. Will do some additional testing using just the virtual printer instead of bambu printer plugin to see if it's related or not shortly. 

```
2025-03-15 18:01:21,080 - octoprint.server.api.files - ERROR - There was an error retrieving the last modified data from storage sdcard and path Cabel_Clip_Hook.gcode.3mf
Traceback (most recent call last):
  File "C:\OctoPrint-Windows\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint\server\api\files.py", line 87, in _create_lastmodified
    return fileManager.last_modified(
  File "C:\OctoPrint-Windows\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint\filemanager\__init__.py", line 1076, in last_modified
    return self._storage(location).last_modified(path=path, recursive=recursive)
  File "C:\OctoPrint-Windows\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint\filemanager\__init__.py", line 1080, in _storage
    raise NoSuchStorage(f"No storage configured for destination {location}")
octoprint.filemanager.NoSuchStorage: No storage configured for destination sdcard
```
this is the information sent via virtualized sd card list from my plugin. 
> Recv: cabel_~1.3mf 308126 0x5a6f8e21 "Cabel_Clip_Hook.gcode.3mf"

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

I will test in safe mode, just haven't yet since I was working with a virtualized file list

### Version of OctoPrint

1.11.0rc4

### Operating system running OctoPrint

Windows 10 64 Bit

### Printer model & used firmware incl. version

Bambu

### Browser and version of browser, operating system running browser

Chrome Version 134.0.6998.89 (Official Build) (64-bit)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250315183724.zip](https://github.com/user-attachments/files/19266912/octoprint-systeminfo-20250315183724.zip)

will test with virtual printer and it's fake sd card shortly and provide feedback.

jneilliii, 2025-03-15:
Issue is reproducible for me with the virtual printer and running in safe mode. Error occurs when a file is selected from sd card and the octoprint.log gets spammed with error. Here's updated system info bundle. 

[octoprint-systeminfo-20250315193012.zip](https://github.com/user-attachments/files/19267150/octoprint-systeminfo-20250315193012.zip)

jneilliii, 2025-03-15:
the logging spam could be a front-end bug because it seems to get in a request loop. 

[image]

foosel, 2025-03-17:
Can reproduce and confirm. Couldn't you have run into this a week earlier? XD (/jk, thanks for reporting!)

foosel, 2025-03-17:
Pretty sure that's a side effect of #5073

jneilliii, 2025-03-17:
That was my assumption as well.

foosel, 2025-03-17:
Fix ready for 1.11.0rc5

foosel, 2025-03-18:
1.11.0rc5 is out

jneilliii, 2025-03-23:
hmm...not sure if this is fixed. I just tested with my Bambu plugin, which use file_tree_extension to add 3mf support, and appears to still be getting into a request loop.

[image]

[octoprint-systeminfo-20250323115519.zip](https://github.com/user-attachments/files/19409954/octoprint-systeminfo-20250323115519.zip)

plugin related?

jneilliii, 2025-03-24:
possibly red herring and is a plugin issue. with virtual printer and a 3mf file it doesn't do this. more than likely related to `Can't create a collision free filename`.

closed 2025-03-24 by jneilliii