OctoPrint/OctoPrint #5122: [RC Feedback] Feedback on the 1.11.0rc5 Release Candidate

foosel, 2025-03-18:
Please provide *general* feedback on your experience with the 1.11.0rc5 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. If you run into any bugs however, [**please open a full bug report**](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) and link it here! Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- The newly added Custom Control Manager plugin.
- The newly added Health Check plugin.
- The newly added Upload Manager plugin.
- The MFA-TOTP plugin using the new MFA plugin interface.
- Timelapse creation, especially with a configured rendering delay.
- Remember me functionality

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles* in the issue tracker, not just a statement somewhere that something doesn't work! 
###
### Please respect the process and don't make the already very stressful RC phase any more difficult for me!
###
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

jneilliii, 2025-03-22:
not sure if it's a regression or a pre-existing issue, but reauthentication (possibly autologin related) is a little weird. https://github.com/OctoPrint/OctoPrint/issues/5125

b-morgan, 2025-03-24:
Updated RPi 3B, OctoPi 0.18, Python 3.7.3, RPi Camera v2, LulzBot TAZ 6 from rc4 to rc5.

Updated RPi 4B, OctoPi* 1.0.0cam (build 2023.07.20.144556), Python 3.9.2, RPi Camera v3, Virtual Printer from rc4 to rc5.

foosel, 2025-03-26:
I have fixed the reauthentication issue (which was also present in 1.10.3 but way less likely to trigger, and it was actually a logic bug), and I also found a bug in the error tracking plugin and some log spam from sentry, which I also both fixed.

I thought about pushing a new RC today, but I rather give this one a bit more time for testing before @jneilliii finds yet another weird bug and then I have to roll out rc7 for that :P 

And for the record, I'm actually happy about everything that's found now! I'm just trying to reduce the overhead of pushing out RCs here.

ckuethe, 2025-03-31:
Seems to be working OK with my Ender 5S1:
- homeassistant can poll my octopi over API
- octodroid can control my octopi over API
- SuperSlic3r can upload prints over API
- Web UI is working
- ArcWelder plugin works correctly
- (abandoned) PrettyGCode plugin still displaying correctly
- Auto Bed Level plugin working
- "Camera Streamer" streaming correctly
- Print is, well, printing.

foosel, 2025-04-01:
New release candidate 1.11.0rc6 is out, feedback ticket is here: #5129

closed 2025-04-01 by foosel