OctoPrint/OctoPrint #5123: Mangling G2/G3 command arguments

gbarbehenn, 2025-03-19:
### The problem

Slice and print, from Cura 5.9 via Octoprint to Elegoo Neptune 3 Pro, the attached project.
Printer reports: "Error G2/G3: bad parameters" for:
"N69 G3 X.221.02.233 Y202.746 E53.67545*70"

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

I'm printing to Octoprint from Cura 5.9.  How do I start Octoprint in "safe" mode in this scenario?

### Version of OctoPrint

Octoprint 1.10.3/ Python 3.10.5

### Operating system running OctoPrint

Windows 10 Home Version 22H2 Installed on 5/‎5/‎2024 OS build 19045.5608 Experience Windows Feature Experience Pack 1000.19061.1000.0 (with Thinstuff remote desktop client) 

### Printer model & used firmware incl. version

Elegoo Neptune 3 Pro Firmware v2.1.2.3b/TFT v1.4.2

### Browser and version of browser, operating system running browser

v136.0.2 (64-bit)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [x] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

If gcode file is printed from SD card, it prints correctly.

[G2G3 bug in Octoprint.zip](https://github.com/user-attachments/files/19343899/G2G3.bug.in.Octoprint.zip)

gbarbehenn, 2025-03-19:
Your bug report page doesn't allow *.3mf project files from Cura.  I had to zip it, and just zipped all the files, in the system info bundle into the *.zip file, that I attached.
However, in the interest of collaboration, please find the file reattached, by itself:

[octoprint-systeminfo-20250319082428.zip](https://github.com/user-attachments/files/19343960/octoprint-systeminfo-20250319082428.zip)

jneilliii, 2025-03-19:
> I'm printing to Octoprint from Cura 5.9. How do I start Octoprint in "safe" mode in this scenario?

Receiving a file from Cura whil OctoPrint is running in safe mode should work as it normally does, you just have to select Restart OctoPrint in Safe Mode from the system menu. The file transfer from Cura side is handled with core functionality of OctoPrint. 

[image]

It would help if you were to also enable serial logging in settings to diagnose printer communication.

foosel, 2025-03-20:
Also share the gcode please, as I'd like to rule out that this line is actually in there, in which case this would be a Cura bug.

foosel, 2025-04-22:
@gbarbehenn Could you please reply to the questions you were asked? We need information from you. Unless you respond, this will have to be closed.

gbarbehenn, 2025-04-28:
As I posted above, all the requested 'additional' information is the zip file I originally attached  Including the system info bundle zip file.  The bug post page is very persnickety about what files it will and won't allow, specifically *.3MF files are allowed.  So, I just bundle everything together, gcode files, error screen captures, etc, into one zip file and upload.  I did also send the sytem info bundle zip separately.
If bug reporting allowed uploading of *.3mf or pdf files, I would't have to zip them.  But, please look in the original uploaded zip file for the files you seek.

jneilliii, 2025-04-28:
The 3mf file you shared was not a sliced 3mf file, it is only the model itself and whatever settings you had enabled. If it was a sliced file, it would also have a plate_1.gcode file inside the Metadata folder inside it.

foosel, 2025-05-12:
We need the gcode that fails to render. Please download it *from OctoPrint*, zip it (I sadly don't have any power over GitHub supporting gcode uploads so this step is sadly necessary) and share it here.