OctoPrint/OctoPrint #5125: [1.11.0rc5]: reauthentication causes reload and full login page

jneilliii, 2025-03-22:
### The problem

When prompted to reauthenticate for opening get more in plugin manager the UI pops a reload overlay and then shows full login page. Not sure if it's related but I do have autoLogin enabled in config. Will do additional testing with that disabled. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0rc5

### Operating system running OctoPrint

OctoPi 1.0.0

### Printer model & used firmware incl. version

Virtual

### Browser and version of browser, operating system running browser

Google Chrome Version 134.0.6998.118 (Official Build) (64-bit)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [x] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

you'll have to catch the js error from the video hopefully. because it does the reload so fast and flushes out the console. 

https://github.com/user-attachments/assets/78f8d175-1085-4b41-baf9-b446d8f45879

[octoprint-systeminfo-20250322142851.zip](https://github.com/user-attachments/files/19404910/octoprint-systeminfo-20250322142851.zip)

jneilliii, 2025-03-22:
not related to autologin, disabled that on my instance and issue is still reproducible. testing with remember me option checked now.

jneilliii, 2025-03-22:
remember me option checked made no difference.

foosel, 2025-03-24:
The good news is, I can reproduce this. The bad news is, the only error I see is related to a font file. Will start to dig. Thanks for finding that!

foosel, 2025-03-24:
I can also reproduce this under 1.10.3 as long as I set a break point at the right place. So apparently this is an existing bug, though something changed between 1.10.x and 1.11.0 enough that it now triggers immediately.

The problem is here: 

https://github.com/OctoPrint/OctoPrint/blob/d79a0d20f3f1c7f2edb56dedda3b70267a937e65/src/octoprint/server/api/__init__.py#L311-L312

This will cause the user to get logged out if they already have a login session. This should NOT get triggered obviously when the user is reauthenticating, however it currently is. The only reason why it works in 1.10.3 is that for some reason then the `user` object is not a `SessionUser` instance, which at a later line will cause this logout to get stopped. However, if I set a breakpoint at this line, it IS a `SessionUser`. And I currently have no clue what kind of race condition is causing this behaviour, but it obviously is one.

I have a solution (not doing the logout when the current user is the same as the login is coming from, as in, it's a reauthentication), but I'd prefer to fully understand first why this ever worked in the first place before I go with that...

jneilliii, 2025-03-24:
Yeah, I thought I had seen this in the current stable release version before, and could never nail down what was going on. I think it might happen more with UI Customizer installed possibly, but never really was able to isolate when it was and wasn't happening.

foosel, 2025-03-26:
Fix ready for 1.11.0rc6

foosel, 2025-04-01:
1.11.0rc6 is out

commit 194dd29c (2025-03-25): fix: don't log out on reauth

Not a regression, was also present in 1.10.x, but due to some
interesting race condition was not as easily reproduced there.

Closes #5125
closed 2025-04-01 by foosel