OctoPrint/OctoPrint #5127: Folder paths don't work

gbarbehenn, 2025-03-26:
### The problem

Create folder paths as "ip address\whatever\uploads" in Octoprint-settings-folders.  Note that "\\ip address" and "\\NASname" don't work, only "ipaddress\..." or "NASname\...".
This path can now be 'tested' successfully, as seen in the attached PDF.
After use, checking this path is found to be empty?
Checking the host\Octoprint install directory path, new folders are found ...\ipaddress\whatever\uploads.
Even thought path test succeeds, it is actually testing for the presence of the path on the host *AND* these aren't used anyway.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Octoprint server, on host no longer responds, see attached PDF.  I did wait 0.5 hours or so, and close/reopen the browser window to ipaddress:5000.

### Version of OctoPrint

Octoprint 1.10.3 Python 3.10.5

### Operating system running OctoPrint

Both host and client are running Windows 10 Home 64b.  Host has Thin Stuff Remote Desktop Client

### Printer model & used firmware incl. version

Elegoo Neptune 3 Pro firmware v2.1.2.3b, TFT v1.4.2

### Browser and version of browser, operating system running browser

Firefox 136.0.3

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

It appears to me that the Octoprint-settings-folder *cannot* even imagine a directory that isn't on the local host.  This isn't a huge problem for Gcode files which are a few tens of megabytes, but is a huge problem for video.  My current printjob will run just over 25 hours.  The webcam video for this is already tens of gigabytes.
Also, not only are *all* folder settings in Octoprint-settings-folder made (and tested) on the local machine, but they are in the Octoprint install directory.  In Windows, this path could include spaces (although Octoprint itself must install in the root, not "~\Program Files\", where it should be, for the same reason.  This isn't a 'silly' point, Windows has special protections for Program Files or Program Files (x86), and has had since Windows XP SP2, see Windows Virtual Authentication) which Octoprint cannot handle at all.
The browser can certainly handle either of these contingencies, as you can force it to ask for download destination, and then give it any path that makes you happy and it will succeed.

[Attempt to restart Octoprint in safe mode.pdf](https://github.com/user-attachments/files/19471172/Attempt.to.restart.Octoprint.in.safe.mode.pdf)
[Octoprint setting-folders.pdf](https://github.com/user-attachments/files/19471174/Octoprint.setting-folders.pdf)
[octoprint-systeminfo-20250326090119.zip](https://github.com/user-attachments/files/19471173/octoprint-systeminfo-20250326090119.zip)

jneilliii, 2025-03-26:
Have you tried using mklink (windows symlink equivalent) to point a specific local folder path to the remote location?

cp2004, 2025-03-26:
> Also, not only are all folder settings in Octoprint-settings-folder made (and tested) on the local machine, but they are in the Octoprint install directory.

By default it tries to use AppData for this, but you can override it via the command line when you run OctoPrint - have you specified anything here?

I'd agree with @jneilliii - using the IP directly seems prone to issues, you could setup your network storage on your windows server so it is mapped to a drive letter, and then use this letter (like X:\OctoPrintUploads) to specify the path you want rather than using the IP. If it has a domain name, on windows you could setup to use '\\server.domain.com\OctoPrintUploads' for example.

On your OctoPrint Server machine (but not within OctoPrint itself - as if you accessed it with Remote Desktop), which method do you use to access the network storage?

jneilliii, 2025-03-27:
Good point @cp2004 you can set the basedir parameter with the OctoPint command, and the windows installer I made includes an option to set that path during the install process.

foosel, 2025-04-22:
@gbarbehenn Could you please reply to the questions you were asked? We need information from you. Unless you respond, this will have to be closed.

closed 2025-05-12 by foosel