OctoPrint/OctoPrint #5129: [RC Feedback] Feedback on the 1.11.0rc6 Release Candidate

foosel, 2025-04-01:
Please provide *general* feedback on your experience with the 1.11.0rc6 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. If you run into any bugs however, [**please open a full bug report**](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) and link it here! Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- The newly added Custom Control Manager plugin.
- The newly added Health Check plugin.
- The newly added Upload Manager plugin.
- The MFA-TOTP plugin using the new MFA plugin interface.
- Timelapse creation, especially with a configured rendering delay.
- Remember me functionality

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles* in the issue tracker, not just a statement somewhere that something doesn't work! 
###
### Please respect the process and don't make the already very stressful RC phase any more difficult for me!
###
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

fonso2616, 2025-04-01:
remember me option on login screen seems to not be working again.  I have to re-login every so often

jneilliii, 2025-04-01:
> remember me option on login screen seems to not be working again. I have to re-login every so often

❗👇 IMPORTANT 👇❗
If you run into any obvious bugs not yet listed below, please open a new ticket and follow "How to file a bug report". I need full bug reports with system info bundles in the issue tracker, not just a statement somewhere that something doesn't work!
Please respect the process and don't make the already very stressful RC phase any more difficult for me!
❗👆 IMPORTANT 👆❗

Is it from the same browser/client machine that you see this. Remember me is per browser/client.

foosel, 2025-04-02:
What @jneilliii said. "It doesn't work" doesn't allow me to do anything - it works here, it's now part of the automatic end-to-end test suite which also reported green on this ever since fixing the original issue, and the code responsible for it hasn't even been touched either. So if something isn't working, I need a proper bug report including logs, browser information, ideally detailed reproduction steps. I really want to iron out any glaring bugs here, but I *need help* to do that. I can't do it based on "it doesn't work", at least not if it works here.

It should be noted that the Remember Me functionality has always and still does work by the use of a `remember_me` cookie, so if that gets deleted, and the session times out, yes, you will be logged out. Some browser settings might be overly aggressive in that regard, and there's nothing I can do about that. If I can't rely on the cookie store and local storage functioning as advertised, things will misbehave as a consequence.

jneilliii, 2025-04-03:
Seems login session isn't sticking and getting prompted to login after some unknown timeframe. Someone else reported on Discord and I've verified reproduction. Issue opened: https://github.com/OctoPrint/OctoPrint/issues/5132

fonso2616, 2025-04-03:
K, thanks for the update.

mine started doing it intermittently again also.

I can't find a pattern or series of events to duplicate, it just randomly looses the remember_token_P443  and an active_logout_P443      true gets create in the cookies

Fonso
817-366-3195

foosel, 2025-04-03:
@fonso2616 I also took the liberty to edit your comment to remove your phone number and also purged that from history. Be advised that any reply via email to a notification from an issue to GitHub will end up as a response on the issue from your account, *including any mail signature*. So I'd really recommend to either not reply to GitHub notifications on issues (a good idea in any case as that usually will cause broken formatting and/or attachments to vanish in the ether without any sign of them ever having been there), or to remove the phone number in your signature unless you want to have it in the open.

foosel, 2025-04-03:
1.11.0rc7 is out, new ticket is #5133

closed 2025-04-03 by foosel