OctoPrint/OctoPrint #5131: [Request] Add "sensors" category to REST data model for use by plugins

n8many, 2025-04-03:
### The problem

I am using the Enclosure plugin and have some sensors that read temperature and humidity of the printer. The plugin currently just feeds these sensors to the 'temperature' data for the printer, which has the side effect of these values appearing as "temperature" regardless of what they really are. This solution does work within the current framework

When plugins like PlotlyTempGraph and external applications like Home Assistant or OctoApp pull this data, they think it is native Celcius when it is really a %RH. It's a mild annoyance.

### Proposed solution

What if to /api/printer we added a "sensors" section? This would be a section where any plugin could post data for sensors that would be accessible by other plugins and API users. 

The data format could look something like 

```
{
...
  'sensors': {
    'ambient temperature': {  # name set by plugin
      'actual': 23.4,         # for consistency with temperature api
      'type': 'temperature',  # from list: [temperature, humidity, light_level, switch, power, air_quality, other]
      'unit': 'C'             # from valid list for each sensor type, no restriction on other.
    },
    'ambient humidity': {
      'actual': 26.5,
      'type': 'humidity', 
      'unit': '%'
    }
  }
...
}
```

This would allow multiple plugins to publish to the same API. This way, it is easier for plugins to take over being a data source if an old plugin gets deprecated, or to combine multiple data sources that support old and new sensors. This would also incentivize apps/plugins to support extra sensor data since it is now coming from a centralized source and they don't have to re-implement it for every plugin trying to read sensor data.

### Alternative solutions

Obviously, I could modify the plugin or ask the app/plugin maintainers who use the "temperatures" data to also support the rest API for the Enclosure plugin, however, this is a plugin that hasn't been maintained for a few years and if someone comes along and writes a new version they'll have to reimplement that support.

Alternatively, 'type' and 'unit' could be added to "temperatures", but that feels a bit like a hack. It would remove the need for other plugins support a second section of API and you would not suddenly have data disappear if a sensor source migrated to the "sensors" section, and the sensor display app/plugin did not update. It feels more appropriate to me to leave the "temperatures" sections for things that have an "actual" and "setpoint", but I could see not wanting two places where that information is stored.

### Additional context

I have poked around the Enclosure plugin enough that I think I could add support for this (and maybe pull in a few of the lingering pull requests), though I am not looking to become a full time maintainer.

jneilliii, 2025-09-28:
> just display these non-temperature values differently

That's what the `Sub` check box is for in my plugin.

[image]

[image]

aerickson, 2025-09-28:
Thank you! Sorry for missing that option.

Seems to be working great. 

My only grumble is that they're labeled C or F... I may hack on a toggle like the Fahrenheit one to handle 'raw' values.

[image]

aerickson, 2025-09-29:
> I got a fix for that too. You can modifying config.yaml directly, as the setting is not exposed to the front-end settings dialog. Adjust the hover_template property.

Awesome. Working perfectly. 

Thank you! 😄