OctoPrint/OctoPrint #5132: [1.11.0rc6]

jneilliii, 2025-04-03:
### The problem

After letting interface sit for a while, not sure what the timing is, prompted for reauthentication. Seems sneakily similar to previous issue #5125 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0rc6

### Operating system running OctoPrint

OctoPi 1.0.0

### Browser and version of browser, operating system running browser

Google Version 134.0.6998.179 (Official Build) (64-bit)

### Printer model & used firmware incl. version

VIRTUAL

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

https://github.com/user-attachments/assets/2da39d06-bbcf-4632-908c-56d87da979f8

[octoprint-systeminfo-20250402230119.zip](https://github.com/user-attachments/files/19579054/octoprint-systeminfo-20250402230119.zip)

foosel, 2025-04-03:
For some reason the session becomes stale. Usually that should not happen as any kind of action including the open websocket should keep it alive, but something's broken there apparently and the session ages out, leading to a log-out.

The idea behind this mechanism is to kill sessions for which the browser tab got closed or otherwise lost, to reduce the likelihood of abuse of stolen session cookies. The default session timeout is 15min, and that matches the time span in your systeminfo bundle from active login to session deletion and logout perfectly. I'm now going to make this configurable so I can easier test and debug this.

jneilliii, 2025-04-03:
If it makes any difference, I had minimized the browser during testing.

foosel, 2025-04-03:
Found it already.

It turns out that `threading.Thread.is_alive` is a method, NOT, as I had assumed due to its name, a property. And when I fixed an issue that showed up due to the fix of #5125 (because two wrongs made one right here) I put in this check into the `on_login` handler of the server side websocket handler:

``` python
if not self._keep_alive.is_alive:
    self._keep_alive.start()
```

This was necessary to prevent double starts due to reauth without the (wrong) logout beforehand. Alas... this will always be `False` and thus the keep alive will never get started, because `is_alive` is a bound method, and that's a true-thy value. This makes it all work:

``` python
if not self._keep_alive.is_alive():
    self._keep_alive.start()
```

foosel, 2025-04-03:
Fix ready for 1.11.0rc7, which I'm trying to roll out today so we don't miss the weekend for community testing.

foosel, 2025-04-03:
1.11.0rc7 is out

commit 4e64e485 (2025-04-03): fix(auth): is_alive is a method, not a property

Fixes #5132