OctoPrint/OctoPrint #5133: [RC Feedback] Feedback on the 1.11.0rc7 Release Candidate

foosel, 2025-04-03:
### Feedback

Please provide *general* feedback on your experience with the 1.11.0rc7 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. If you run into any bugs however, [**please open a full bug report**](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) and link it here! Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- The newly added Custom Control Manager plugin.
- The newly added Health Check plugin.
- The newly added Upload Manager plugin.
- The MFA-TOTP plugin using the new MFA plugin interface.
- Timelapse creation, especially with a configured rendering delay.
- Remember me functionality

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles* in the issue tracker, not just a statement somewhere that something doesn't work! 
###
### Please respect the process and don't make the already very stressful RC phase any more difficult for me!
###
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

### Verification

- [x] I'm a maintainer or collaborator of OctoPrint and allowed to use this issue type

b-morgan, 2025-04-03:
Updated RPi 4B, OctoPi* 1.0.0cam (build 2023.07.20.144556), Python 3.9.2, RPi Camera v3, Virtual Printer from rc5 to rc6, ran into the bug in #5132, and updated from rc6 to rc7.

Updated RPi 3B, OctoPi 0.18, Python 3.7.3, RPi Camera v2, LulzBot TAZ 6 from rc5 to rc7.

fonso2616, 2025-04-04:
All is well now in my world with 1.11.0rc7. sessions stayed logged in all day yesterday
thanks for fixing it so quickly !!

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

closed 2025-04-22 by foosel