OctoPrint/OctoPrint #5134: USB Webcam doesn't work

classicrocker883, 2025-04-04:
### The problem

I used to be able to use my Logitech QuickCam Pro 9000 without issue on Raspberry Pi Zero 2 W running OctoPi/OctoPrint.
Now, it simply doesn't seem to work. I made sure the camera **does** work on another machine.
This started after I updated to a newer version of OctoPrint and upgraded the files (`sudo apt-get upgrade).

I believe this may have to do with `mjpg-streamer` because it just isn't available now.
The farthest I'm able to get it working is the "Control" tab where the webcam shows is like trying to load with a black screen. I tried going to the webpage IP (127.0.0.1), however this may be a discrepancy since my actual OctoPrint local page IP (10.0.0.126).

I tried a few things by editing the boot config files as it was suggested, like defining the resolution and FPS, but no.

Separately, I even tried this on a non-raspberry pi, the Radxa Rock-3C. I installed OctoPrint without issue, and t the webcam does work like using "Cheese" or through a terminal, except it isnt recognized in OctoPrint.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

OctoPi

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

I will upload required info

classicrocker883, 2025-04-07:
update: uploaded systembundle info

foosel, 2025-04-07:
> This started after I updated to a newer version of OctoPrint and upgraded the files (`sudo apt-get upgrade`).

The `apt-get upgrade` might have broken something. From the looks of it, your webcam server isn't even running (its log is completely empty), so something broke there and it's impossible to say from just the logs. Logging into the system via SSH and running `systemctl status webcamd` and `journalctl -u webcamd` might tell you more about what's wrong, but that's not a guarantee.

Please note though that the webcam server is external to OctoPrint and thus this bug tracker is the wrong location to get help with issues like this. I recommend you move to either the community forum or the discord server, which are the official support channels. See https://help.octoprint.org for links to both. Feel free to reference this ticket.

As this is not something actionable on my end, I'm closing this issue now.

closed 2025-04-07 by foosel

classicrocker883, 2025-04-09:
thanks for the update. I suppose what I can try is a fresh install, might this fix it? I'm guessing raspberry pi cameras should work without issue, this might just be USB related, or perhaps just something to do with me?

foosel, 2025-04-10:
If the above commands don't unearth any helpful log messages, then yes, a fresh install would most likely be the easiest solution. You should be able to just backup/restore - OctoPrint's backup doesn't include the full system state but only its own configuration and data, and that should play no roll here.

classicrocker883, 2025-04-12:
GOOD NEWS! I got it working.

I didnt reinstall a new version, instead I manually installed `mjpg-streamer`

```bash
git clone https://github.com/jacksonliam/mjpg-streamer.git
cd mjpg-streamer/mjpg-streamer-experimental
sudo make
sudo make install
```