OctoPrint/OctoPrint #5138: Downstream dependency limits version 5.0.0 breaks

Hillshum, 2025-04-17:
### The problem

Getting this error on startup of a new install using `octoprint_deploy`

```2025-04-17 14:52:04,026 - octoprint.startup - INFO - ******************************************************************************
2025-04-17 14:52:04,029 - octoprint.startup - INFO - Starting OctoPrint 1.10.3
2025-04-17 14:52:04,030 - octoprint.startup - INFO - Starting in SAFE MODE. Third party plugins will be disabled!
2025-04-17 14:52:04,031 - octoprint.startup - INFO - Reason for safe mode: problem during last startup
2025-04-17 14:52:04,031 - octoprint.startup - INFO - ******************************************************************************
2025-04-17 14:52:04,487 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2025-04-17 14:52:04,514 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2025-04-17 14:52:04,514 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2025-04-17 14:52:04,514 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2025-04-17 14:52:04,525 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2025-04-17 14:52:04,541 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/plugins, /home/pi/.tester/plugins and installed plugin packages...
2025-04-17 14:52:08,002 - octoprint.plugin.core - INFO - Found 20 plugin(s) providing 20 mixin implementations, 43 hook handlers
2025-04-17 14:52:08,074 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2025-04-17 14:52:08,219 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/OctoPrint/bin/octoprint", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/__init__.py", line 1008, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/click/core.py", line 1697, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/cli/server.py", line 317, in serve_command
    run_server(
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/cli/server.py", line 163, in run_server
    octoprint_server.run()
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/server/__init__.py", line 389, in run
    self._setup_app(app)
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint/server/__init__.py", line 1577, in _setup_app
    limiter = Limiter(
              ^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/flask_limiter/extension.py", line 280, in __init__
    self.init_app(app)
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/flask_limiter/extension.py", line 342, in init_app
    raise ConfigurationError(
limits.errors.ConfigurationError: Invalid rate limiting strategy fixed-window-elastic-expiry
```

Looks like the newly released verion 5.0.0 of the `limits` package removed support the `fixed-window-elastic-expiry` strategy https://limits.readthedocs.io/en/stable/changelog.html

Running `pip install limits==4.7.2` from within the OctoPrint venv fixed the issue.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

Raspberry Pi OS bookworm

### Browser and version of browser, operating system running browser

Edge

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250417150055.zip](https://github.com/user-attachments/files/19799949/octoprint-systeminfo-20250417150055.zip)

peddaahh, 2025-04-18:
Can confirm an some others too
https://community.octoprint.org/t/another-octoprint-will-not-start/63031

CybrGost69, 2025-04-18:
if you follow this [guide](https://3dprintscape.com/install-octoprint-on-linux/) after the install and while the virtual environment is still available, run the `pip install limits==4.7.2` then add the user. worked for me

foosel, 2025-04-18:
Will tackle this with 1.11.0 (due Tuesday). Just weird it doesn't show on the automatic test install and server startup twice daily.

foosel, 2025-04-22:
> Just weird it doesn't show on the automatic test install and server startup twice daily.

Rate limiting is disabled on CI to allow for E2E testing of "wrong password" behaviour. Same for my local dev environment. That's why it didn't break there.

Note to self: create basic test in CI with enabled rate limiting and CSRF

---

1.11.0 will ship with `limits` pinned to `<5` *and* the rate limiting strategy set to `fixed-window`. Only the latter would in theory suffice already, but there are some more breaking changes in limits 5 and given that its release was only days ago, I'll rather err on the side of caution here and give things an upper pin now that reflects the versions available during the RC phase.

commit a3369462 (2025-04-22): chore(deps): pin limits to <5

limits 5.0.0 released just days ago and contains breaking changes. As we are about to
roll out a new OctoPrint release and limits 5 has already caused issues with 1.10.x in the
field, it'll be pinned for now to reduce the likelihood of further issues.

Closes #5138