OctoPrint/OctoPrint #5142: 404 loop with incorrectly selected SD card file

jneilliii, 2025-04-25:
### The problem

While working on the Bambu plugin and auto-upload/start from slicer I ran into a 404 request loop with a long file name file. I know this is a weird bug related to how the plugin is doing something wrong, but as requested on Discord creating a tracking issue in case it's reproducible in some other random way. 

### Proposed solution

Fix my plugin to not select a long file name in code with self._printer.select_file and instead use it's 8.3 equivalent. Would be curious if Klipper does something similar with long file name support or not and might impact your comm shimming efforts. 

### Additional information

[octoprint-systeminfo-20250424195418.zip](https://github.com/user-attachments/files/19900641/octoprint-systeminfo-20250424195418.zip)

### Verification

- [x] I'm a maintainer or collaborator of OctoPrint and allowed to use this issue type

jneilliii, 2025-04-25:
I think this might not be a loop after all. It might be my plugin trying to identify the printing file that isn't selected, because of the long filename possibly at this line, which will try with every MQTT payload update from printer. 

https://github.com/jneilliii/OctoPrint-BambuPrinter/blob/rc/octoprint_bambu_printer/printer/states/printing_state.py#L86

foosel, 2025-05-13:
@jneilliii have you been able to confirm whether that is only caused by your plugin? I so far haven't seen it again.

closed 2025-05-13 by jneilliii