OctoPrint/OctoPrint #5143: Babel for v1.11 requires python 3.8

VinnyAmbesek, 2025-04-25:
### The problem

Can't update to octoprint v1.11
```
Collecting Babel<2.15,>=2.14.0
Downloading https://www.piwheels.org/simple/babel/Babel-2.14.0-py3-none-any.whl (11.0 MB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
Babel<2.15,>=2.14.0 from https://www.piwheels.org/simple/babel/Babel-2.14.0-py3-none-any.whl#sha256=a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185 (from OctoPrint==1.11.0):
Expected sha256 a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185
Got        36b6870daee6f00356902ac31f0ae3e2005581cd08f2f2a36c3e22e7c8f7cdd2

The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

Trying to install through pip using ssh gives the error:

```
pi@octopi:~ $ ~/oprint/bin/pip install --no-cache-dir babel==2.16
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Ignored the following versions that require a different python version: 2.15.0 Requires-Python >=3.8; 2.16.0 Requires-Python >=3.8; 2.17.0 Requires-Python >=3.8
ERROR: Could not find a version that satisfies the requirement babel==2.16 (from versions: 0.9.6, 1.0, 1.1, 1.2, 1.3, 2.0, 2.1.1, 2.2.0, 2.3.0, 2.3.1, 2.3.2, 2.3.3, 2.3.4, 2.4.0, 2.5.0, 2.5.1, 2.5.2, 2.5.3, 2.6.0, 2.7.0, 2.8.0, 2.8.1, 2.9.0, 2.9.1, 2.10.0, 2.10.1, 2.10.2, 2.10.3, 2.11.0, 2.12.0, 2.12.1, 2.13.0, 2.13.1, 2.14.0)
ERROR: No matching distribution found for babel==2.16

```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.3

### Operating system running OctoPrint

OctoPi 0.18.0

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250425000732.zip](https://github.com/user-attachments/files/19902213/octoprint-systeminfo-20250425000732.zip)

jneilliii, 2025-04-25:
I mislead you in discrod when we were working on this. I should have had you try this command for python 3.7.

```
~/oprint/bin/pip install --no-cache-dir babel==2.14.0
```

jneilliii, 2025-04-25:
When running the above command @VinnyAmbesek got the same original error.

```
pi@octopi:~ $ ~/oprint/bin/pip install --no-cache-dir babel==2.14.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting babel==2.14.0
  Downloading https://www.piwheels.org/simple/babel/Babel-2.14.0-py3-none-any.whl (11.0 MB)
     ━━━━━━━╺━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.0/11.0 MB 166.3 kB/s eta 0:00:55
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    babel==2.14.0 from https://www.piwheels.org/simple/babel/Babel-2.14.0-py3-none-any.whl#sha256=a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185:
        Expected sha256 a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185
             Got        4bddc8e81c769856f01a4bdbbc0962393c1e081726508a7aea2e21426451c51d
```

jneilliii, 2025-04-25:
Similar issue on the forum (https://community.octoprint.org/t/update-to-1-11-0-fails/63098?u=jneilliii) pointed to pihole, which we don't believe was the cause here. PIhole was disabled and pi rebooted while disabled to make sure. Not even clear if it's pihole or possible piwheels related issues.

foosel, 2025-04-25:
Whatever this is, it is completely outside of my control. 

Something's going wrong when fetching this dependency from the network causing this mismatch. This is an error I can't reproduce on my own network. 

My money is still on something like pihole or maybe some provider shenanigans messing with the traffic. Or maybe piwheels is having issues. Apparently interrupted downloads will cause hash errors, so it might also just be a flaky connection: https://github.com/pypa/pip/issues/11153

Nothing actionable from OctoPrint's end here, this is a problem that needs to be solved a layer beneath it.

closed 2025-04-25 by foosel

foosel, 2025-04-25:
Just to document this, it doesn't seem to be a general issue with piwheels either:

``` bash
$ curl -O https://www.piwheels.org/simple/babel/Babel-2.14.0-py3-none-any.whl                                                                                                                                                                                                                                           % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 10.5M  100 10.5M    0     0  1717k      0  0:00:06  0:00:06 --:--:-- 1538k
$ sha256sum Babel-2.14.0-py3-none-any.whl 
a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185  Babel-2.14.0-py3-none-any.whl
```

I also tested updating from various versions to 1.11.0 on OctoPi 0.18 with Python 3.7, successfully:

[image]

and addition to that have also successfully updated my long running instance attached to my MK3 that's still on 0.18 and Python 3.7 (yes, I need to update myself).

I just now also flashed a fresh 0.18.0-latest on a Pi3 and ran through the update - zero issues. Whatever this is - it's not a general issue, it's probably a network problem, or maybe some ad blocker being a bit too aggressive (though I run pihole on my lan myself with quite some aggressive block lists). Note that in general with a network wide ad blocker I'd suggest to explicitly put things like `github.com` and `piwheels.org` on the whitelist.

That being said, the only change I could *imagine* has an impact here is c5a57432a84b023c94671b8c2a6fa9fb8aa009de. However, that would not explain why data with the wrong hash is being downloaded from piwheels (which is what is happening based on the above error). The hash that pip expects is actually the correct one. I only could imagine this change to have an impact if pip was expecting the `wrong` hash here.

Anyhow... even IF this change is indeed the issue here for some people for some reason, a bugfix release will have to wait. I need a break, it's either that now for a week, or I'll be gone for months soon. 

If all else fails, this should work (on OctoPi, adjust venv as needed for manual installs):

``` bash
curl -O https://www.piwheels.org/simple/babel/Babel-2.14.0-py3-none-any.whl
~/oprint/bin/pip install Babel-2.14.0-py3-none-any.whl
```

Manually downloads the wheel through curl and tells pip to install it. If you are feeling funny you can also check the `sha256sum` of the downloaded whl first:

```
sha256sum Babel-2.14.0-py3-none-any.whl
```

Should be `a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185`

spanzetta, 2025-04-28:
I did follow the suggestion and the hash mismatch was solved.. but at the very end of the update I got the following other errors..

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pydrive2 1.15.4 requires google-api-python-client>=1.12.5, which is not installed.
pydrive2 1.15.4 requires oauth2client>=4.0.0, which is not installed.
pydrive2 1.15.4 requires pyOpenSSL>=19.1.0, which is not installed.

How could I solve?

jneilliii, 2025-04-28:
Those are specific to Google drive related plugins I'm pretty sure. You could attempt the same pip install command with the module and versions listed.

spanzetta, 2025-04-28:
Yes.. thanks.. sorted out using command from console
pip install google-api-python-client oauth2client pyOpenSSL

and then again the SW update from Octoprint interface (this time successfully)
Thanks

Kjubyte, 2025-08-28:
Hi,
I found this issue through Google, so I want to share my findings with others. I had the same issue after setting up a new OctoPi instance. To me, it looks like an issue with the piwheels server.

The domain resolves to multiple IPs:
```
$ host www.piwheels.org
www.piwheels.org is an alias for proxy.mythic-beasts.com.
proxy.mythic-beasts.com has address 93.93.129.174
proxy.mythic-beasts.com has address 46.235.225.189
proxy.mythic-beasts.com has IPv6 address 2a00:1098:0:82:1000:3b:1:1
proxy.mythic-beasts.com has IPv6 address 2a00:1098:0:80:1000:3b:1:1
```

I tried to download the file from each server by specifying the IP address to curl and noticed that I have problems with only 2/4 IPs. In my case, 46.235.225.189 and 2a00:1098:0:82:1000:3b:1:1 are working fine and are fast (download in <1s), but the other two are **very** slow, and I get multiple connection errors while trying to download the file.

To test the download with a specific IP, use:
```
curl -v -O https://www.piwheels.org/simple/babel/babel-2.16.0-py3-none-any.whl --resolve 'www.piwheels.org:443:[2a00:1098:0:80:1000:3b:1:1]'
```

To use a known working server you can force usage of a specific IP by adding an entry to the hosts file. Alternatively, wait a few minutes and try again. OctoPi will randomly select an IP, so your next try might just work.

foosel, 2025-08-29:
Interesting! Looping in @bennuttall as that's maybe something he might be interested in looking into?

bennuttall, 2025-08-29:
Can confirm the file hash is correct:

```
piwheels@piwheels:~/www/simple/babel$ sha256sum Babel-2.14.0-py3-none-any.whl
a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185  Babel-2.14.0-py3-none-any.whl
piwheels@piwheels:~/www/simple/babel$ grep Babel-2.14.0-py3-none-any.whl index.html 
<a href="/simple/babel/Babel-2.14.0-py3-none-any.whl#sha256=a30fdf4ee308b7d13b3bbb6a1ce7b598be958c355d4ce0347e603b2b63eac185" data-requires-python="&gt;=3.7">Babel-2.14.0-py3-none-any.whl</a><br>
```

We cover this problem in our FAQ: https://piwheels.org/faq.html

We recommend manually downloading and installing the file as others have suggested.

> I tried to download the file from each server by specifying the IP address to curl and noticed that I have problems with only 2/4 IPs. In my case, 46.235.225.189 and 2a00:1098:0:82:1000:3b:1:1 are [...]

@Kjubyte interesting! These aren't different servers, just different routes to the same server. I'll check with our hosts.

Kjubyte, 2025-08-29:
@bennuttall I did the tests again today. It's better now, but 46.235.225.189 and 2a00:1098:0:82:1000:3b:1:1 are still working fine every time, while the other two now mostly work, but sometimes they don't.

If they work, the download is often fast (1MB in <1s). If they don't, it takes minutes. I had a few cases where I could download the file, ran the test again, and now the download is very slow and aborts after some time (seconds to minutes, very inconsistent).

I ran a mtr and saw that sov.proxy.mythic-beasts.com (93.93.129.174, 2a00:1098:0:80:1000:3b:1:1) has quite a lot of packet loss (up to 20%), while cll.proxy.mythic-beasts.com (46.235.225.189, 2a00:1098:0:82:1000:3b:1:1) has 0%.

Maybe that helps identify the problem.

bennuttall, 2025-08-29:
Which country are you in?

FYI you can set the specific IP in your `/etc/hosts` file