OctoPrint/OctoPrint #5144: [Request] Add a disable to the "Delete File" confirmation prompt

11Below, 2025-04-25:
### The problem

Please allow me to disable the "Delete File" confirmation prompt.  When I do accidently delete a file I wasn't going to, I just upload it again from the source.

### Proposed solution

Add a switch to allow me to disable the "Delete File" confirmation prompt.

### Alternative solutions

Remove the "Delete File" confirmation prompt.

jneilliii, 2025-04-25:
Feature was just added in 1.11.0 to appease https://github.com/OctoPrint/OctoPrint/issues/833. Adding a printDeletedConfirmation option to <https://docs.octoprint.org/en/master/configuration/config_yaml.html#feature> isn't a bad idea for those that want the old ways.

11Below, 2025-04-25:
Yeah, I should have added a smiley face at the end of my Alternate Solution.

jacopotediosi, 2025-04-30:
This could also be an opportunity for a plugin, to allow users to enable/disable the feature from the web, without having to edit the file config.yaml.

We could take inspiration from this plugin, which already alters the confirmation behavior: https://plugins.octoprint.org/plugins/inlineconfirm/

fly74, 2025-05-11:
Plz make it configurable.

foosel, 2025-05-12:
Adding yet another config flag was *precisely* what I wanted to avoid and why I only added the confirmation after adding a bulk delete option via the now bundled upload manager:

[image]

[image]

Every option means more crowded config, more things to migrate/maintain/etc. But ok then, option added, ready for 1.11.1.

jefflorenzini, 2025-05-12:
I think you were being naive if you added a delete confirmation without an option to turn it off. I appreciate you adding the flag, thank you.

foosel, 2025-05-12:
Explain to me please why you require being able to turn it off. Not asking to be annoying, I actually want to understand the use case I missed here. 

The idea was to make it less easy to accidentally delete a file on the small file manager on the side, while allowing full blown bulk management including bulk deletion through the upload manager. From how *I* use things, that would work well for me. From how I understood plenty of others used it, it would work well too. So apparently, some people are using things differently here, and it would help a lot with future decisions to understand how. 

So please, explain.

fly74, 2025-05-12:
In my case it is not necessary to having a print library in the file browser. Sometimes I print a lot and don't care about deleting last prints. Later then I delete while scolling some files wich I think I don't need anymore. If I delete a file accidentally it is not a problem, I can easily create it from the source/slicer. So in the past without prompt it was perfect for me. But I understand your intention. Was the prompt discussed before ?

foosel, 2025-05-12:
The prompt was requested in #833 (originally in a nice way, then things turned nasty). 

The clean-up use case was supposed to be better supported by the now included upload manager (which I hope everyone chiming in here has actually tried?), as that offers a bulk delete, and multi select, which makes it *way* more usable for that task.

jefflorenzini, 2025-05-12:
Here's the workflow: slice in PrusaSlicer, upload to Octoprint, then print. Once it's printed, if I figure I'll never print that particular thing again, I'll delete it. I don't want a confirmation dialog because if I accidentally delete something, I can just re-slice and upload again from PrusaSlicer. 

The idea is that Octoprint always has a _copy_ of the thing I'm printing. I save my projects in PrusaSlicer, so if I need to print it again I can open that project and just reslice and upload and print.

Another workflow is when I'm printing something I designed in Fusion 360. In that case, I'll have many test prints and I'll be uploading and printing and deleting them as I go along so I don't end up with 5 prints all named the same and 4 of them were older versions of my object.

My point is, deleting the wrong object is not a concern, as I can just reslice. Having to confirm deletion is just annoying, so that's why I want the option to turn it off.

jneilliii, 2025-05-12:
> Your particular use case actually screams for a plugin that would offer to delete the file for you after it finishes printing, or even just outright do that without asking, at least on a successful print.

This plugin exists already: <https://plugins.octoprint.org/plugins/DeleteAfterPrint/>

fly74, 2025-05-12:
> Ok. It's not a concern for you, but it was a concern for others,

That's why we ask for the ability to disable it per option because there were 2 use cases.

jneilliii, 2025-05-12:
> I have just been too lazy to request that the "Upload" buttons get moved to the top of the files area.

You know you can configure most soccer's to send the gcode file directly to OctoPrint and you wouldn't even have to use the upload button; or drag the file into the Web interface for the upload overlay and drop it on the left.

foosel, 2025-05-13:
> most soccer's

I'm pretty sure that was supposed to say "slicers"... auto correct claims another victim ^^

> This plugin exists already: https://plugins.octoprint.org/plugins/DeleteAfterPrint/

You definitely have a better overview of the plugin repository these days than I, thank you! 👍 

---

And thanks everyone for the feedback. I learned about another way that apparently plenty of people use OctoPrint that so far I didn't know about, simply because feedback I so far got and use I saw didn't cover that at all.

You should know that every change I do on things like this is always to make things better for everyone, while also keeping things secure, maintainable and usable. The latter means that every setting I add, and especially those that I also expose on the UI, gets a ton of thought on whether it actually is needed, because I don't want to overload the settings with stuff most people don't require. That's also part of my responsibility as a maintainer, and due to a lack of information this time I made the wrong call. As already say this will be fixed as part of 1.11.1, and in fact I'll start the release process today.

foosel, 2025-05-13:
1.11.1 has been released.