OctoPrint/OctoPrint #5145: 1.11.0 unable to generate application keys due to reauthentication

jneilliii, 2025-04-28:
### The problem

1. Disable reauthentication in config.yaml.
```
accessControl:
  defaultReauthenticationTimeout: -1
```
2. Try to generate application key
3. 403 forbidden error with response that doesn't prompt anything...

[image]

Saw similar issue on another instance that doesn't have reauth disabled and from settings it wouldn't pop the password prompt, but from the user settings > application keys tab it did, and after worked in regular application keys settings too. Tried to reproduce this and couldn't so possible race condition somewhere? 

### Proposed solution

fix reauth logic for application keys plugin to not error when disabled

### Additional information

reauth disabled instance:

https://github.com/user-attachments/assets/239e6f34-f86f-4a3f-8e91-32d4d3e9de36

### Verification

- [x] I'm a maintainer or collaborator of OctoPrint and allowed to use this issue type

jneilliii, 2025-04-28:
Issue reproducible in safe mode.

jacopotediosi, 2025-04-29:
The check is performed in the following lines of code and affects not only appkeys generation, but all backend endpoints that require reauthentication:
https://github.com/OctoPrint/OctoPrint/blob/e8cb5c4bbd8e93c0002c3fd7db63b53e3b1aa9ad/src/octoprint/server/util/flask.py#L1674-L1677

According to [the documentation](https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control):

>   \# Default timeout after which to require reauthentication by a user for dangerous changes, in minutes.
>   \# Defaults to 5 minutes. Set to 0 to disable reauthentication requirements (SECURITY IMPACT!).
>   defaultReauthenticationTimeout: 5

Instead, you are setting `defaultReauthenticationTimeout` to the invalid value `-1`, which in Python evaluates to True and causes the unexpected behavior described in your issue.

I suggest modifying the line:
```minutes = settings().getInt(["accessControl", "defaultReauthenticationTimeout"])```
to
```minutes = settings().getInt(["accessControl", "defaultReauthenticationTimeout"], min = 0)```

This change would ignore negative values and enforce a non-negative timeout, aligning with the documented behavior and intent.

jneilliii, 2025-04-29:
Thanks for pointing that out, I think I read the docs wrong on this one, it should be set to 0, not -1. Not sure where I got -1 from

closed 2025-04-29 by jneilliii