OctoPrint/OctoPrint #5147: stuck on "Opening serial connection" when connecting the printer

AbdElhalim12, 2025-04-29:
### The problem

```
Changing monitoring state from "Offline" to "Opening serial connection"
Connecting to port /dev/ttyUSB0, baudrate 115200

```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0

### Operating system running OctoPrint

octopi 1.0.0

### Printer model & used firmware incl. version

creality cr10

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250429075843.zip](https://github.com/user-attachments/files/19953653/octoprint-
systeminfo-20250429075843.zip)

this issue start happening after I updated octorpint to 1.11.0
I have to restart the system more than once to connect to the printer successfully.
I don't have backup of octoprint

jacopotediosi, 2025-05-01:
@AbdElhalim12 your systeminfo is missing serial logs. Please enable them via Settings > Serial Connection > Log communication to serial.log and update the systeminfo bundle you attached to this issue.

foosel, 2025-05-12:
Sadly all that `serial.log` shows is a successful connect - I need a `serial.log` from an *unsuccessful* attempt to connect to further debug this. @AbdElhalim12 can you provide this please?

foosel, 2025-05-12:
Might also be a duplicate of #5151, or rather, sharing the core reason.

foosel, 2025-05-13:
Chances are high this is either a duplicate of #5151 or #5149, both of which have been fixed in 1.11.1. So please, upgrade to that and check if the issue persists. If it does persist, then please make sure `serial.log` is enabled, reproduce the issue, immediately (!) create a system info bundle and post that here, requesting a reopen.

closed 2025-05-13 by foosel