OctoPrint/OctoPrint #5149: Octoprint 1.11.0 stuck on "Connecting" when connecting to Prusa MK3S with uBlock origin active

piranha32, 2025-05-01:
### The problem

After upgrade to Octoprint 1.11.0 from 1.10.(?) I can no longer connect to the printer. The serial port opens properly, there is active communication between printer and Octoprint, but the connection never advances to "Connected" state.
Log of serial port communication during connection attempt (closed manually, it can go on and on indefinitely): 
[2025-04-30_octoprint_prusa_serial_comms.txt](https://github.com/user-attachments/files/20001491/2025-04-30_octoprint_prusa_serial_comms.txt)

I tried:
- Restart Octoprint
- Restart the host computer
- Restart the printer
- Restart the printer and the host computer
- Restart Octoprint in Safe Mode
- Disabling uBlock Origin

*Sometimes* connection succeeds in safe mode with uBlock Origin disabled, but I could not identify the combination of operations to make it reproducible.
Screenshots of browser console and network traffic with uBlock disabled attached.

[image]
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

N/A

### Version of OctoPrint

1.11.0

### Operating system running OctoPrint

Build 2023.07.20.144556 with "camera-streamer", based on OctoPi 1.0.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

### Browser and version of browser, operating system running browser

Firefox, Fedora 41

### Printer model & used firmware incl. version

Prusa MK3S, Printer firmware version: 3.13.2-7080, MMU firmware version: 3.0.1

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [x] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250430121325.zip](https://github.com/user-attachments/files/20001498/octoprint-systeminfo-20250430121325.zip)

jneilliii, 2025-05-01:
Will probably need you to enable serial logging in OctoPrint's settings and attempt again so we can see the full communication stack. Also looks like your situation is more complicated because of an MMU2 in the mix as well? One thing that stands out right away opening the system info bundle in bundleviewer is...

[image]

what kind of power supply are you using (phone chargers aren't power supplies) and/or additional USB peripherals do you have plugged into the pi?

piranha32, 2025-05-01:
Re-uploading system info with serial log. 
The power supply is CanaKit DCAR-052A5, 5V/2.5A. The only connected peripherals are the camera and the printer. 
I ordered a 3.5A power supply, but up until the upgrade the system was stable (except occasional gentle throttling during timelapse processing, which did not affect functionality in any way).

[octoprint-systeminfo-20250501130156.zip](https://github.com/user-attachments/files/20002830/octoprint-systeminfo-20250501130156.zip)

piranha32, 2025-05-09:
Downgrading Octoprint to 1.10.3 restored functionality.

jds11111, 2025-05-12:
I am also having problems connecting to the Creality Ender 3 S1. It gives this in the error logs. Will also try rolling back Octoprint to something pre 1.11.0.
```
Traceback (most recent call last):
  File "/home/jay/oprint/lib/python3.9/site-packages/octoprint/util/comm.py", line 3222, in _monitor
    self._perform_detection_step()
  File "/home/jay/oprint/lib/python3.9/site-packages/octoprint/util/comm.py", line 3515, in _perform_detection_step
    self._trigger_error(error_text, "autodetect")
  File "/home/jay/oprint/lib/python3.9/site-packages/octoprint/util/comm.py", line 4123, in _trigger_error
    self._callback.on_comm_error(
  File "/home/jay/oprint/lib/python3.9/site-packages/octoprint/printer/standard.py", line 1545, in on_comm_error
    self._error_info = ErrorInformation(
  File "/home/jay/oprint/lib/python3.9/site-packages/pydantic/main.py", line 253, in __init__
    validated_self = self.__pydantic_validator__.validate_python(data, self_instance=self)
pydantic_core._pydantic_core.ValidationError: 2 validation errors for ErrorInformation
faq
  Input should be a valid string [type=string_type, input_value=None, input_type=NoneType]
    For further information visit https://errors.pydantic.dev/2.11/v/string_type
logs
  Input should be a valid list [type=list_type, input_value=None, input_type=NoneType]
    For further information visit https://errors.pydantic.dev/2.11/v/list_type
```

foosel, 2025-05-12:
That's another issue though, see #5151. Make sure the printer is actually connected and showing up in the dropdown *before* hitting connect.

commit 07f59687 (2025-05-12): fix(comm): always send hello on seeing start

Will solve an issue with connecting to printers when start is seen after
sending the initial hellos. Logic error crept in via 2ce52c09

Fixes #5149