OctoPrint/OctoPrint #5151: Attempting connection without serial port results in deadlock (no future connection possible)

oerkel47, 2025-05-04:
### The problem

When there is no serial port detected (printer unpowered) it is still possibl to attempt a connection. Doing so will result in octoprint communication module stuck in error.
> State: Error
>No more candidates to test, and no working port/baudrate combination detected.

Even if you get a valid serial port later (printer switched on and octoprint detected the port) the error mode is  stuck and connecting is blocked.  I believe this must be related to  octoprint 1.10 making com errors more persistent(?).

A workaround could be to reset the error once the port detection system has detected a port - or block connection attempts when no port is ready. It would be great if a  "port is available and ready" variable would be exposed to plugins to better handle connecting. This problem can be emphasized with psu control plugin and insufficient power on delay before an automatic connection attempt. 

I found this works for now:
 ` is_ready_for_connect = (self._printer.get_connection_options()['ports'] and self._printer.get_state_string() not in ("Detecting serial connection","Opening serial connection","Connecting"))`

Curiously I only noticed this after Octoprint 1.11 Release. The necessary delay might have changed for whatever timing reason between octoprint versions or it was just by chance.

Additionally I have encountered the error on the screenshot which is related to the com error code.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0

### Operating system running OctoPrint

OctoPi 1.0.0

### Browser and version of browser, operating system running browser

Firefox

### Printer model & used firmware incl. version

Prusa Mini Prusa-Firmware-Buddy 6.1.2+7894 (Github)

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

https://github.com/user-attachments/assets/f1dee5e9-295a-40ac-9eab-b6786f3ea588

[octoprint-systeminfo-20250504194447.zip](https://github.com/user-attachments/files/20028886/octoprint-systeminfo-20250504194447.zip)

[image]

jneilliii, 2025-05-04:
Looks like your octoprint.log may have rolled over and it's missing the error in your screenshot.

oerkel47, 2025-05-04:
True, I  reset the log before to show what happens when I took the screen recording as it was full of my own debugging plugin stuff. The error in the attached screenshot only happens sporadically and is probably not directly related to the reason for this report which is the port being blocked. I have taken a look at the referenced code though and it appears some variables are "None" under certain conditions which triggers the validationError.

oerkel47, 2025-05-04:
Another interesting find:

1. A disconnect command (even though not actually connected) can bring the printer from error to offline.
2. Then if a serial port is available, it is possible to start a connecting process. This will fail though and create the error from the screenshot (log below)

**Bold** are annotations by me to clear things up

> 2025-05-04 21:31:29,687 - octoprint.plugins.psucontrol - INFO - Switching PSU Off 
  **DISCONNECT command is send by psu control here**
2025-05-04 21:31:29,688 - octoprint.plugins.psucontrol - DEBUG - Switching PSU Off Using GPIO: 21
2025-05-04 21:31:29,690 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline"
**State goes from Error to offline**
2025-05-04 21:31:29,725 - octoprint.server - INFO - Autorefresh of serial port list stopped
2025-05-04 21:31:29,728 - octoprint.server - INFO - Starting autorefresh of serial port list
2025-05-04 21:31:29,740 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2025-05-04 21:31:29,841 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2025-05-04 21:31:29,841 - octoprint.plugins.psucontrol - DEBUG - Result: False
2025-05-04 21:31:29,842 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2025-05-04 21:31:29,842 - octoprint.plugins.psucontrol - DEBUG - PSU state changed, firing psu_state_changed event.
2025-05-04 21:31:30,770 - octoprint.server - INFO - Serial port list was updated, refreshing the port list in the frontend
2025-05-04 21:31:34,843 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2025-05-04 21:31:34,844 - octoprint.plugins.psucontrol - DEBUG - Result: False
2025-05-04 21:31:34,844 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2025-05-04 21:31:39,846 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2025-05-04 21:31:39,847 - octoprint.plugins.psucontrol - DEBUG - Result: False
2025-05-04 21:31:39,847 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2025-05-04 21:31:44,851 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2025-05-04 21:31:44,852 - octoprint.plugins.psucontrol - DEBUG - Result: False
2025-05-04 21:31:44,852 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2025-05-04 21:31:49,855 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2025-05-04 21:31:49,855 - octoprint.plugins.psucontrol - DEBUG - Result: False
2025-05-04 21:31:49,856 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False
2025-05-04 21:31:53,923 - octoprint.server - INFO - Serial port list was updated, refreshing the port list in the frontend
2025-05-04 21:31:54,859 - octoprint.plugins.psucontrol - DEBUG - Polling PSU state...
2025-05-04 21:31:54,859 - octoprint.plugins.psucontrol - DEBUG - Result: False
2025-05-04 21:31:54,860 - octoprint.plugins.psucontrol - DEBUG - isPSUOn: False 
**Port is available and I can click connect**
2025-05-04 21:31:55,913 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Detecting serial connection"
2025-05-04 21:31:55,930 - octoprint.server - INFO - Autorefresh of serial port list stopped
2025-05-04 21:31:55,935 - octoprint.util.comm - INFO - Serial detection: Performing autodetection with 1 port/baudrate candidates: /dev/ttyACM0@115200
2025-05-04 21:31:55,935 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyACM0, baudrate 115200
2025-05-04 21:31:55,936 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
2025-05-04 21:31:55,975 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #1 with timeout 2.0s
2025-05-04 21:31:55,987 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:55,991 - octoprint.util.comm.command_phases - DEBUG - phase: queued | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:55,995 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:56,006 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2025-05-04 21:31:56,012 - octoprint.util.comm.command_phases - DEBUG - phase: sent | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:57,172 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #2 with timeout 2.0s
2025-05-04 21:31:57,175 - octoprint.util.comm.command_phases - DEBUG - phase: queuing | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:57,180 - octoprint.util.comm.command_phases - DEBUG - phase: queued | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:57,182 - octoprint.util.comm.command_phases - DEBUG - phase: sending | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:57,185 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2025-05-04 21:31:57,200 - octoprint.util.comm.command_phases - DEBUG - phase: sent | command: M110 N0 | gcode: M110 | tags: [ trigger:comm.say_hello, trigger:detection ]
2025-05-04 21:31:57,253 - octoprint.util.comm - INFO - Serial detection: Handshake attempt #3 with timeout 2.0s
[... cut ...]
  Input should be a valid list [type=list_type, input_value=None, input_type=NoneType]
    For further information visit https://errors.pydantic.dev/2.11/v/list_type
2025-05-04 21:31:57,511 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_crash, payload: {'commerror_text': 'See octoprint.log for details', 'throttled_now': False, 'throttled_past': True, 'throttled_mask': 917504}
2025-05-04 21:31:58,752 - octoprint.util.comm - INFO - Changing monitoring state from "Error" to "Offline after error"

oerkel47, 2025-05-12:
Thank you for investigating and fixing @foosel!

foosel, 2025-05-13:
I'm just glad it was such an easy fix ^^ I have to fix up some small stuff here (new laptop, test env is not fully there yet) and then I can start the release process. Might even be able to push out 1.11.1 still today.

commit 46633549 (2025-05-12): fix(comm): ErrorInformation fields need to be optional

Closes #5151
closed 2025-05-13 by foosel