OctoPrint/OctoPrint #5152: [Request] recommended third party plugins on first boot

jonnytest1, 2025-05-06:
### The problem

started with octoprint a few days ago and for every plugin i needed i basically found out i needed it one print to late 😅 (exaggerated)

### Proposed solution

so i was wondering about a popup on first start that gives a list of like the top 10 recommend plugins (with a warning about them being third party)

including things like:
Power Failure Recovery
Print history plugin
Octolapse

(and probably some i still dont know about :/)

jneilliii, 2025-05-06:
I doubt this will happen because what is recommend for one person might not be recommended for others. Plugin Manager actually has a sort method for most installed plugins, and I think that's the default sort order. 

[image]

You can also see on the [plugin repo](https://plugins.octoprint.org/) website a list on the right.

[image]

foosel, 2025-05-12:
This is not something I'll implement, *especially* not after [having the stats manipulated in favor of two competing plugins](https://octoprint.org/blog/2024/07/04/more-stats-manipulation/) last summer. Next thing I'll have to explain to plugin authors why plugin A was put on that list and not plugin B, or if generated from the stats we'll see even more manipulation.

The plugin repository is there for a reason, where needed the FirmwareCheck plugin will tell you about plugins that you need for your printer to work, and that's where we'll leave it.

closed 2025-05-12 by foosel