OctoPrint/OctoPrint #5153: Disabling connectivity check has no effect

alithechemist, 2025-05-08:
### The problem

Enable regular connectivity check  is disabled
Enable plugin blacklist processing on startup  is disabled

and yet:

```
2025-05-08 14:46:02,913 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2025-05-08 14:46:02,914 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2025-05-08 14:46:02,914 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is 

```
also from systeminfo:

```
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: True
connectivity.enabled: False
connectivity.online: True
```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.0

### Operating system running OctoPrint

gnu+Linux 

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250508145336.zip](https://github.com/user-attachments/files/20103249/octoprint-systeminfo-20250508145336.zip)

alithechemist, 2025-05-08:
[image]

foosel, 2025-05-12:
Patch ready for 1.11.1

commit 0a173905 (2025-05-12): ux(connectivity): clearly state when the checker is disabled

The log should not contain lines that look like it's enabled in such a
case.

Fixes #5153