OctoPrint/OctoPrint #5154: [Request] For reasons like [5153] Connectivity check should be disabled by default

alithechemist, 2025-05-08:
### The problem

Users privacy should come first

### Proposed solution

Disable connectivity check by default

### Alternative solutions

Remove connectivity check and implement error messages in case the users tries to download an update without connectivity

### Additional context

No application should ever phone-home or connect to third parties services without explicit users consent.
Failing doing  this compromises the privacy of the users.

foosel, 2025-05-12:
Disabling the connectivity check by default will lead to most of OctoPrint's user base never enabling it. Thus, it won't help with what it was created for - seeing issues with the user's network and not trying to get updates etc due to that. 

The connectivity check is basically just "try to open this port on this host" and the default is a DNS server. So the privacy impact of it is as minimal as possible. This approach was chosen over just checking whether there's a default route defined on the network as that won't tell us if that route is actually working or not.

For privacy reasons, the connectivity check is supposed to support disabling it (defaulting to "I'm online"), and also supports changing the host & port for connection to something else. A bug in this is not a reason to make the whole feature completely useless for its purpose - trying to make sensible decisions for less technical users who don't know how to determine whether their system is even online before opening support issues about things with online requirements not working.

I'll look into what's up with #5153

closed 2025-05-12 by foosel

AlexDashwood, 2025-05-15:
> No application should ever phone-home or connect to third parties services without explicit users consent

For your information, when you first launch an Octoprint, the first thing you see is a setup wizard. The setup wizard asks you to set the initial values, one of which is the Connectivity check. Until you explicitly choose to enable or disable it, you will not be able to proceed to the next step and use the octoprint accordingly. If I'm wrong, please correct me.

AlexDashwood, 2025-05-29:
Tried to experiment a little:
1. Clean install of OctoPrint. The Setup Wizard shows up.
2. Get to the point where it asks about Connectivity check.
3. Tried to hit "Next" without choosing to enable or disable the function. It didn't allow me to continue.
4. You also can't close or quit this; there's no "exit" or "X" button.
5. Tried to reload the page. It shows the Setup Wizard again and asks to choose the state of Connectivity check.
6. Tried to manually restart the OctoPrint service. It shows the Setup Wizard again and asks to choose the state of Connectivity check.

So, basically, you can't get past the Setup Wizard, at least without any manual hacks. The only possible way I see is to restore your previous installation by uploading a backup in the Setup Wizard. That way, you don't get to a point to choose Connectivity check. However, to get a backup, you need a working installation in the first place, where you also need to make a decision to enable or disable Connectivity check. So, there's no way to "phone-home" without explicit consent.

P.S. Sorry for my English; I'm not a native speaker.