OctoPrint/OctoPrint #5156: access_validation_factory fails to raise 403, returns 500 error

jacopotediosi, 2025-05-15:
### The problem

I was implementing authentication for a route (`octoprint.server.http.routes`) in a plugin, inspired by the implementation in the "backup" plugin:

https://github.com/OctoPrint/OctoPrint/blob/70a82678c3ae78ff5187c347a92d220017327c27/src/octoprint/plugins/backup/__init__.py#L370-L406

It seems to me that [permission_validator](https://github.com/OctoPrint/OctoPrint/blob/70a82678c3ae78ff5187c347a92d220017327c27/src/octoprint/server/util/flask.py#L1518) and [permission_and_fresh_credentials_validator](https://github.com/OctoPrint/OctoPrint/blob/70a82678c3ae78ff5187c347a92d220017327c27/src/octoprint/server/util/flask.py#L1534) should raise a 403 exception if the user’s permissions are not as expected:

https://github.com/OctoPrint/OctoPrint/blob/70a82678c3ae78ff5187c347a92d220017327c27/src/octoprint/server/util/flask.py#L1518-L1531

However, if I try to download a backup file with wrong privileges (e.g., a user in the Operator group) or without a session, I get a 500 error:

[image]

Even setting the logs to DEBUG level, the only error I see is:
```
2025-05-15 04:37:15,421 - tornado.access - ERROR - 500 GET /plugin/backup/download/octoprint-backup-20250504-155644.zip (127.0.0.1) 1.00ms
```

### Proposed solution

The API should return 403 instead of 500.
I don’t know what causes this bug, but I have confirmed it happens both with `permission_validator` and `permission_and_fresh_credentials_validator`.

### Verification

- [x] I'm a maintainer or collaborator of OctoPrint and allowed to use this issue type

jacopotediosi, 2025-05-15:
I investigated this bug.
This behavior should have already been fixed by commit 81ae1b4, but the fix is likely ineffective.

In [access_validation_factory](https://github.com/OctoPrint/OctoPrint/blob/70a82678c3ae78ff5187c347a92d220017327c27/src/octoprint/server/util/tornado.py#L1848), validators raise `tornado.web.HTTPError` exceptions, but these are not handled specifically. As a result, they fall into the generic `except Exception as e:` block, which causes an HTTP 500 error to be raised:

https://github.com/OctoPrint/OctoPrint/blob/70a82678c3ae78ff5187c347a92d220017327c27/src/octoprint/server/util/tornado.py#L1869-L1876

foosel, 2025-05-15:
Have this scheduled for next week, but this sounds reasonable, thank you for the investigation, that will save some time in fixing this.

foosel, 2025-05-19:
Should be fixed by the above commit, ready for 1.11.2.

commit 1682d6c5 (2025-05-19): fix: return http errors by validators

Closes #5156