OctoPrint/OctoPrint #5157: [Request] use one library to get network interface information

kohend, 2025-05-16:
### The problem

Currently utils/net.py uses the netifaces module, while zeroconf is using ifaddr, for the usage in net.py, ifaddr has all needed features.
I want to add a feature to ifaddr to work on Android devices (currently both netifaces and ifaddr fail on PermissionDenied), adding it to one library would be simpler, and will allow both OctoPrint and Home Assistant to run on Android (via Octo4a/Termux)

### Proposed solution

use the ifaddr.netifaces API instead of netifaces, it's required by Zeroconf anyway.

### Alternative solutions

The alternative solution would be for me to patch net.py and ifaddr/zeroconf to add an alternative means to obtain the info, while not a huge amount of work, it's about double.

### Additional context

I'm a developer and willing to make and test the PR that changes this.

jacopotediosi, 2025-05-19:
Could this be a duplicate (or related) to #5104?

kohend, 2025-05-19:
I think related, I couldn't find it when I searched.
I made the ifaddr PR, BTW.

foosel, 2025-05-19:
Well spotted, as this is a dependency anyhow and netifaces doesn't give us anything else than what it provides here, I've refactored the affected `octoprint.util.net` functions to use `ifaddr` instead. Will be part of 1.12.0.

I *might* still backport it to a 1.11.x bugfix release, but I'm a bit reluctant right now as it is a bit of a non-minimal change.

kohend, 2025-05-19:
Thanks, since this is waiting for a change in ifaddr to be accepted and released to change things in practice, so 1.12.0 should work IMO.

commit 30c7d892 (2025-05-19): refactor: get rid of netifaces dependency in favor of ifaddr

As mentioned on #5157, we already have a third party dependency on
ifaddr anyhow, which can do everything we require from netifaces,
so let's get rid of the dependency on netifaces.

Closes #5157

Might also positively influence #5104