OctoPrint/OctoPrint #5158: CLI add user with --admin no longer working

paukstelis, 2025-05-17:
### The problem

It appears that the CLI command: `octoprint user add myuser --password password --admin` no longer functions. `user add --help` still shows it as an option.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Not relevant

### Version of OctoPrint

1.11.1

### Operating system running OctoPrint

Linux

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Systembundle is not applicable.

jacopotediosi, 2025-05-19:
Hmm, everything seems to be working fine on my end, using OctoPrint 1.11.1 (latest commit on master):

[image]

@paukstelis If you're referring to the new user not showing up in the Access Control section of the web UI, I've noticed it appears only after restarting OctoPrint. I'm not sure if that's the best behavior, but that's what I'm seeing.

Could you clarify what specific issue you're encountering?

jneilliii, 2025-05-19:
In our discord discussion it seemed it was adding the user, but not assigning the admin group. I was unable to reproduce on OctoPi 1.1.0 64 bit with OctoPrint 1.11.1.

paukstelis, 2025-05-19:
> In our discord discussion it seemed it was adding the user, but not assigning the admin group. I was unable to reproduce on OctoPi 1.1.0 64 bit with OctoPrint 1.11.1.

Correct. The user is created, but is not assigned admin and thus has no permissions.
This was reported by someone using Mint and I also confirmed using Ubuntu 24.

foosel, 2025-05-19:
I can't reproduce this either:

``` yaml
admin:
  active: true
  apikey: null
  groups:
  - admins
  - users
  password: ...
  permissions: []
  roles:
  - user
  - admin
  settings: {}
foosel:
  active: true
  apikey: null
  groups:
  - admins
  password: ...
  permissions: []
  roles:
  - user
  - admin
  settings: {}
```

The only difference that I see between the original `admin` and the just created `foosel` user is that the latter wasn't automatically also added to the `users` group, but as the roles are set correctly that shouldn't make a difference. I successfully logged in with this user too, with admin permissions.

At this point I'd actually prefer a test in safe mode (first post says "not relevant", I think differently, as plugins can basically interfere with everything) and with the default user and group manager plus no HTML based user headers involved (no clue whether this might have been an issue here or not).

Also, a fresh system info bundle that actually contains `octoprint.log` and `tornado.log` with more than 0 bytes might also help.

@paukstelis as you mentioned this coming from a user, I'd also like to either have this user who can reproduce this issue take part in this issue here, or I need someone else to reproduce it. Right now, all I can say is based on my own findings plus those of @jneilliii and @jacopotediosi that things work as they should in both 1.11.1 and 1.11.2.dev.

paukstelis, 2025-05-19:
This is happening as part of octoprint_deploy installation, so there are no additional logs, because no user has logged in to a web interface at all yet. I suspect that might be part of the issue. Perhaps a silent fail if some of the "first run" flags are not set appropriate when the user is created?

paukstelis, 2025-05-19:
Running these series of commands on Ubuntu24 clearly fail:
```python3 -m venv /home/$USER/OctoPrintA
/home/$USER/OctoPrintA/bin/pip install --upgrade pip
/home/$USER/OctoPrintA/bin/pip install wheel
/home/$USER/OctoPrintA/bin/pip install OctoPrint
/home/$USER/OctoPrintA/bin/octoprint --basedir /home/$USER/.testop user add admin --password admin --admin```

foosel, 2025-05-19:
Can't reproduce this with a fresh basedir, but CAN reproduce it when I also create a fresh venv first.

foosel, 2025-05-19:
That's caused by some behaviour change in OctoPrint's third party dependency `Click` with version 8.2.0, released on May 10th.

With Click 8.2.0 I can reproduce, with Click 8.1.8 things still work fine.

You'll probably be able to fix things by doing a `pip install Click==8.1.8` before attempting to create a new user. Will now go through the changelog and try to figure out what happened here,
all I can see right now is that `--admin` is no longer causing the internal flag to become `True`, hence no group will be added.

`octoprint user add --group admin <user>` also works with Click 8.2.0.

foosel, 2025-05-19:
Confirmed bug in Click 8.2.0, apparently supposed to get fixed with 8.2.1, see pallets/click#2894.

I suggest to use either of the two workarounds mentioned above:

- manually downgrade Click to 8.1.8 via `pip install Click==8.1.8` or
- use `--group admins` instead of `--admin`: `octoprint user add myuser --group admin --password mypassword`

until 8.2.1 becomes available. Given that there are workarounds available and a bugfix of the lib already scheduled, I don't think a hotfix pinning Click to <8.2 from my end makes sense here given the release overhead.

paukstelis, 2025-05-19:
Good catch. Checked and confirmed 8.1.8 works. The explicit group definition did not work for me with 8.2.0:
```
/home/$USER/OctoPrintA/bin/octoprint --basedir /home/$USER/.testop user add --group admin admin3 --password admin3
User created:
	admin3
		active: True
		groups: 
		permissions: 
```
Is it `--group admin` or `--group admins`?

foosel, 2025-05-19:
`admins`. I fixed the typo in my first comment, the second on listing both possible workarounds was correct.

commit 9751a45b (2025-05-20): chore: mark broken Click dep

Don't try to pull in Click 8.2.0, that has broken flags. See pallets/click#2894.

Let's see if Click 8.2.1 or OctoPrint 1.11.2 will be out first...

Closes #5158
closed 2025-06-10 by foosel