OctoPrint/OctoPrint #5161: 'Reverse Proxy Test' Page Not Working

David-Kopczynski, 2025-05-25:
### The problem

The `Reverse Proxy Test` page errors out with `Internal Server Error`. Logs show that python throws `AttributeError: 'ReverseProxyInfo' object has no attribute 'model_dump'`.

I am facing this issue with the following steps:
1. Install OctoPrint freshly
2. Configure system in wizard
3. Open `Reverse Proxy Test` page or opening `/reverse_proxy_test` directly

I am not sure if this is partly to blame on my system. However, NixOS does not configure any database etc. and I was only using `services.octoprint.enable = true;` with the newest package.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.1

### Operating system running OctoPrint

Linux: NixOS 25.05

### Browser and version of browser, operating system running browser

Firefox 128.10.1esr

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250525105946.zip](https://github.com/user-attachments/files/20430120/octoprint-systeminfo-20250525105946.zip)

foosel, 2025-05-25:
Can confirm, that's not just your environment. 

Probably some refactoring issue that crept in at some point. 

I'll investigate, add an e2e test and see this gets fixed with 1.11.2

foosel, 2025-05-26:
Ok, I'm pretty sure that's a bug in the nix package:

https://github.com/NixOS/nixpkgs/blob/c596c2f0ce95e3f4853656dec7afd741ab32c89c/pkgs/by-name/oc/octoprint/package.nix#L135

This should definitely not be pointing to pydantic_1 for Python 3.12. I suggest you report this to the package maintainer. In general I'd *strongly* advise against this manual approach by the package maintainer(s) to try to match the defined dependencies and their version ranges - as we can see here that can quickly end up in mismatches.

The core issue should however be solved with the above commit, and I'll roll that out with 1.11.2.

David-Kopczynski, 2025-05-26:
> Would it be possible for you to run a pip freeze | grep -i pydantic in the OctoPrint venv/package?

From the package you correctly referenced (https://github.com/NixOS/nixpkgs/blob/nixos-25.05/pkgs/by-name/oc/octoprint/package.nix) the inputs for pip include `pydantic_1` which is currently version `1.10.21`. I just tested changing this to `pydantic` (with version `2.11.1`) and the 'Reverse Proxy Test' page was correctly shown!

Thanks for the help with this issue! I will forward this to the maintainer :)

commit 554b7cc9 (2025-05-26): fix: wrong BaseModel used for reverse proxy info

Closes #5161
closed 2025-06-10 by foosel