OctoPrint/OctoPrint #5168: [Request] Installing OctoPi using the Raspberry Pi Imager instruction incorrect for WiFi

TroelsL, 2025-06-08:
### The problem

The current instructions at https://octoprint.org/download/#octopi are no longer correct for the latest version of OctoPi.

The SSID and password entered into the CTRL+SHIFT+X menu are not applied. Instead, users need to edit the file `wifi.nmconnection`.

### Proposed solution

Update the instructions.

### Alternative solutions

Update the imager, if possible, to edit `wifi.nmconnection` without manual intervention.

jneilliii, 2025-06-09:
This is an OctoPi issue, not OctoPrint bug. If you're still using that keyboard combo to access the settings I suspect you have older version of pi imager. Try updating to the newest version.

jneilliii, 2025-06-09:
> That file was definitely not altered by the imager as it contained only commented lines.

Yeah, the imager doesn't update that file directly and handles it some other way unknown to me on initial startup, probably through a bash script.

TroelsL, 2025-06-09:
Seeing as it still works for others, I will close the issue. I just tested on an empty SD card, and I still had to manually edit the file. If it props up for others, updating the guide would be helpful.