OctoPrint/OctoPrint #5170: Access request broken when webpage not open during initial request

crysxd, 2025-06-12:
### The problem

With 1.11.2 something seems a bit glitchy with the access request. Sometimes the access request looks like below and the "Allow" button has no effect except dismissing the popup, the client requesting access still receives 202 responses

[image]

In the logs I spotted this:
```
2025-06-12 14:09:16,353 - tornado.access - WARNING - 404 POST /plugin/appkeys/decision/undefined (100.99.20.63) 8.82ms
```

Other times it looks like this and works as expected (this is back to back to the above, same client just retrying):
[image]

I can reproduce this behavior by:

- Opening OctoPrint
- Starting an access request from another client
- The access request is now correctly shown by OctoPrint and works as intended
- Reload the OctoPrint webpage
- The access request is now broken

The same happens when the access request is started and the user then opens OctoPrint

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.11.2 

### Operating system running OctoPrint

Linux

### Browser and version of browser, operating system running browser

Edge / latest

### Printer model & used firmware incl. version

/

### Checklist of files to include below

- [x] Systeminfo Bundle (always include!)
- [x] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [x] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20250612145032.zip](https://github.com/user-attachments/files/20711406/octoprint-systeminfo-20250612145032.zip)

[image]

alanmcruickshank, 2025-08-28:
Heya, just confirming that I'm also experiencing the same issue running 1.11.2 stable. For other users who might be experiencing this (in my case triggered by the home assistant integration), the workaround is to ensure you're logged into the octoprint ui _at the same time_ as the home assistant UI. Even separate tabs was enough to have issues, it required two windows open concurrently for the workaround to work.

Confirmed that in the "happy" case, I'd get the IP address of the requesting sevice, and in the "unhappy" case I'd get the other identifier.

Glad to see that the issue has already been resolved on `main`. Thanks for maintaining the project to such a high standard!

foosel, 2025-09-09:
1.11.3 has been released.

commit 24d7d0e5 (2025-06-16): fix(appkeys): fix request dialog after UI reload

Parameter order was broken

Closes #5170
closed 2025-09-09 by foosel