OctoPrint/OctoPrint #5176: on_shutdown is called too late, network already unavailable

jacopotediosi, 2025-07-17:
### The problem

While implementing `octoprint.plugin.ShutdownPlugin` in a plugin, I noticed that the `on_shutdown()` method is called when the system is already shutting down, and network connectivity is no longer available.

This bug is reproducible by installing the following plugin on a real Raspberry Pi running the latest version of OctoPrintt and OctoPi, and then clicking the *Reboot system* button:
```python
# coding=utf-8
import octoprint.plugin

import requests

class shutdowntestPlugin(octoprint.plugin.StartupPlugin, octoprint.plugin.ShutdownPlugin):
    def on_after_startup(self):
        self._logger.info("Hello World!") # # This correctly appears in octoprint.log
    
    def on_shutdown(self):
        self._logger.info("Goodbye world!") # This does not appear in octoprint.log
        
        try:
            response = requests.get('https://www.example.com/', timeout=10)
            response.raise_for_status()
            with open('/home/pi/.octoprint/a.txt', 'w', encoding='utf-8') as f:
                f.write(response.text)
        except Exception as e:
            with open('/home/pi/.octoprint/a.txt', 'w', encoding='utf-8') as f:
                f.write(str(e))

__plugin_name__ = "shutdowntest"
__plugin_pythoncompat__ = ">=3.7,<4"
__plugin_implementation__ = shutdowntestPlugin()
```

[Click here](https://github.com/user-attachments/files/21296604/shutdowntest.zip) to download a prebuilt zip of the above plugin, installable via the Plugin Manager, provided for convenience to easily reproduce the bug.

After the reboot, the content of `/home/pi/.octoprint/a.txt` on my Raspberry Pi 4B+ is:
```
HTTPSConnectionPool(host='www.example.com', port=443): Max retries exceeded with url: / (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xaf08b8e0>: Failed to establish a new connection: [Errno 101] Network is unreachable'))
```

Please note that in the example above, I wrote to a file instead of using logging, because log entries generated inside `on_shutdown()` do not seem to appear in `octoprint.log` (?).

### Proposed solution

I guess the shutdown sequence is wrong, and `on_shutdown` should be called before initiating the shutdown command.

### Additional information

Environment where I encountered the bug:
- Raspberry Pi 4B+
- OctoPrint 1.11.2
- Python 3.9.2
- OctoPi 1.0.0cam (build 2025.04.22.120926)

### Verification

- [x] I'm a maintainer or collaborator of OctoPrint and allowed to use this issue type

jneilliii, 2025-07-17:
What you are basically looking for is an on_before_shutdown callback I think, that requires completing all registered on_before_shutdown overrides and then initiates the shutdown command?

cp2004, 2025-07-17:
I think when we've looked at this before, determined that in theory it is in the correct place in the sequence but that depending on the environment there are cases it's not triggered. It waits for the command from the OS to trigger shutdown, so that it (in theory) would trigger in any case of closing OctoPrint - whether it's ctrl-c on the command line or via systemctl etc. If you just ran a command based on the button in OctoPrint's UI, you wouldn't cover all these other ways it could be shutdown. It's unfortunate that the network shuts down before the plugin gets called - perhaps there's some more advanced configuration that could be in the service file, that's outside my knowledge.

The above is just from memory from long hours trying to dig into this in 2020/21 - so happy to be proven wrong!

guysoft, 2025-07-17:
Looks like a legit bug.

jacopotediosi, 2025-07-18:
A fix has been merged into OctoPi, and several users have confirmed it resolves the issue. Thank you to everyone who contributed to identifying the root cause.

closed 2025-07-18 by jacopotediosi

guysoft, 2025-07-18:
@jacopotediosi Thanks for your contribution! That bug was hiding for 5 years.